clough42/electronic-leadscrew #174: f280049C/CCS Documentation

tarwy1, 2021-08-20:
Hi, less of an issue with the ELS project, but i am trying to make my own version of this using a different seven segment display driver circuitry and some other different elements, but i am having a lot of trouble finding good documentation on the TI launchpad and code composer studio regarding gpio, spi, pwm, generally interfacing with the hardware etc, is anyone able to help with this?

Thanks

JonWoellhaf, 2021-08-22:
I tried to email this, but the email bounced. Let's see if this works.

See if these documents help. I tried to attach them, but no joy. You should be able to find them by searching their titles.

1) TI Launch Pad Development Kit.pdf

2)  TI Microcontroller Tech Ref Man.pdf

Jan

tarwy1, 2021-08-22:
Hi, tried searching those titles and found a lot of results relating to the MSP430 development boards, and none pertaining to the f280049c board, would you be willing to attach a dropbox/google drive share link? Thanks a lot for helping!

Charlie

JonWoellhaf, 2021-08-23:
I can't attach them to a response and have it go through. Are you on Practical Machinist?Sent via the Samsung Galaxy S8, an AT&T 5G Evolution capable smartphone

clough42, 2021-08-23:
The TI page or the particular launchpad has links to all the docs.  I
worked mostly from the reference document, which is thousands of pages.
Not a lot of tutorials out there like with the Arduino platform.

Turner75, 2021-08-29:
Saw this project a few weeks ago and realized what a difference it would make to threading on my ancient Drummond 1924 M4 Lathe, will remove all those exposed noisy change wheels. I have had lots of automation experience in my thirty years employment in the aerospace industry dating back 30 years and have had a home machine shop, mill, lathe, etc, etc since mid 1960's. Programming experience in assembler, Intel 8085/6, Microchip PIC  basic. 
Struggling up a very steep learning curve with Code Composer IDE and C constructs. Got unit to work finally using original CCS Ver. 9.01. to build/compile/flash early experimental els version with lots of help from ELS Part 10, How to Progam a TI LaunchPad. Later versions of CCS, 1.4, generate lots of errors with the original code. 
Question. How to incorporate latest code updates and build, compile, flash, onto els Launchpad firmware. Any help most welcome.
Graham T  
P.S. Clough42 workshop series one of the best on YT.