clough42/electronic-leadscrew #177: ACT Nema 23 Hybrid Stepper behaves like a impact driver

eglische, 2021-08-29:
Hello!

I've played around now for the better part of 2 days. I came to a point where i could get a smooth action when i used the absolute minimal feed. But everything higher, or even a standard metric 0.75 thread and the stepper motor is hammering away, like crazy. it stalls out at maximum and below a bit of that of course without load, no mater the rpm, even on sub 150rpm on the spindle, it sounds like the little guy grinds away something.

At the moment, there are two things i think is important for troubleshooting, the micro step stetting, and the fact that i use a 4:1 shaft output for the encoder, and a cheap encoder i had laying around, giving me much less resolution of course. BUT the RPM is correct (verified with a tachometer), it's stable and it seams ok.

Steps settings: (giving me 4800 steps for a 3:2 Reduction to the 3mm Lead screw comes out to a 3200 s/r on the lead screw)
`#define STEPPER_MICROSTEPS 8
#define STEPPER_RESOLUTION 600`

Encoder Resolution (yes it is low.... got the wrong encoder, new one is on it's way)
`#define ENCODER_RESOLUTION 30`

**The setup as is accurate within reason. IE. the Encoder translates the correct values and advances the carriage the correct amount per spindle revolution**

Things i tried:
- up the voltage from 36v as per specs to 60v as maximum permissible, change the power supply out
- reducing the steps to full step, half step quarter. (it's a 200s/r@1.8° stepper, it's set now to 16 micro steps effectively, the lowest i would go). The idea of going 1/16 now was, i hoped to get it smoother - nope.
- checking the wiring of course
- checked the coils, got 0.9-1ohm = correspond with other steppers i got working good

Things i am going to try later on:
- setup the stepper as a 4th axies on my mill to see if mach3 gives a smoother step train of signals to verify the stepper is not the cooperate.

If someone has a idea, i am glad for any input.
Thanks

eglische, 2021-08-29:
one note on the side: stepper behaves that way even with no load attached, so it's not that it's too weak or anything....

eglische, 2021-08-29:
i hooked it up to my cnc and configured the stepper driver and the stepper as a 4th axis, it runs perfectly well and smooth and has a lot of force, so i have to exclude that possibility and i suspect the signal train coming from the launchpad is choppy/faulty.

at this point, i am out of my water and need help if i want to conclude this project :(

clough42, 2021-08-29:
Yeah, the encoder resolution is *way* too low.  The ELS firmware doesn't do
any time interpolation between encoder pulses.  It depends on having the
encoder pulses arrive nearly as fast (or faster) than the required output
steps.  With such a low-resolution encoder, it's getting one pulse from the
encoder and trying to generate a large burst of steps to catch up to where
it now thinks it needs to be.  The results are as you observe.

With small discrepancies, it isn't a big deal, because the micro-stepping
absorbs the small bursts of steps.

James

eglische, 2021-08-30:
Good Morning and Thank you :-)

Yes, i was afraid so. Have ordered a used 2000p and a "standard" new 1024p.
Guess i have to gear up the output as well now. to get near that sync ratio (1600 Pulses / 1mm feed Rev) Or/and Go down with the micro stepping.

After sleeping on it, and your confirmation on that, i think this issue can be closed.
Thanks again James

Cheers
Roger

closed 2021-08-30 by eglische