clough42/electronic-leadscrew #178: Higher Threading Pitch

SamGuliver, 2021-09-06:
I wanted to add a higher threading pitch for making customs springs.
I am not very good at coding so my attempt at simply adding missing values to the "metric_thread_table" resulted in multiple errors. Can somebody help me, finding out what I might need to change as well to add higher threading pitches?

Kind Regards

JonWoellhaf, 2021-09-06:
Hi, Sam.

I’ll try to help. Which version of James’ ELS are you trying to modify? The version number is displayed on power up.

Jan

From: SamGuliver 
Sent: September 6, 2021 06:57
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: [clough42/electronic-leadscrew] Higher Threading Pitch (#178)

I wanted to add a higher threading pitch for making customs springs.
I am not very good at coding so my attempt at simply adding missing values to the "metric_thread_table" resulted in multiple errors. Can somebody help me, finding out what I might need to change as well to add higher threading pitches?

Kind Regards

SamGuliver, 2021-09-07:
Hey Jon,

well thanks to you I tried out the newer 1.4 Version which seems to have now resolved my error while compiling and the threading numbers show up correctly when I use try to use them in the UI.

But have to found a different issue that seems to have existed before I did my update as well. The threads don´t go higher than 4mm/rev. I can enter 5 or 6 or higher, but the actual pitch of the resulting thread does not exceed 4mm/rev. Everything below 4 works accurately. Maybe you know whats happening? 
Thank you for helping!

Kind regards

clough42, 2021-09-07:
Can you post the lines you added to the table?  The value to be displayed
and the actual value used for the computations are in different fields.

The compilation errors you were seeing were due to the memory configuration
in the older versions of the firmware.  It was configured to keep the
static data (like the tables) in one bank of memory, and when it grew too
large to fit, it generated an error.  It's now configured to automatically
roll over and use additional memory banks as needed.

James

SamGuliver, 2021-09-08:
Sure. 

 { .display = {BLANK, SIX | POINT,   FIVE,  BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(650) },
 { .display = {BLANK, SEVEN,         BLANK, BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(700) },
 { .display = {BLANK, SEVEN | POINT, FIVE,  BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(750) },
 { .display = {BLANK, EIGHT,         BLANK, BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(800) },
 { .display = {BLANK, EIGHT | POINT, FIVE,  BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(850) },
 { .display = {BLANK, NINE,          BLANK, BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(900) },
 { .display = {BLANK, NINE | POINT,  FIVE,  BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(950) },
 { .display = {ONE, ZERO,           BLANK, BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(1000)},
 { .display = {ONE, ZERO | POINT,   FIVE,  BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(1050)},
 { .display = {ONE, ONE,             BLANK, BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(1100)},
 { .display = {ONE, ONE | POINT,     FIVE,  BLANK}, .leds = LED_THREAD | LED_MM, HMM_FRACTION(1150)},

These are the ones I added to the Tables.cpp file. But as I said I am experiencing issues with the unaltered code as well (above 4mm and above a certain range in the imperial threading too. As I am turning the chuck by hand, the calculations not keeping up, or the stepper being too slow, should not be an issue).  

Thank you.

blackfister, 2023-04-25:
Hi Sam, did you manage to get coarser threads?
If so, the coding posted should be copied into tables, at what line, if I may ask. Thank you

johnsonm, 2023-04-26:
See my PR #255 which does three things:
* Adds, optionally, a lot more thread specifications
* Adds the ability for new thread pitches not supported by the previous configuration (one more decimal digit for both metric and imperial pitches, allowing pitches like .125mm and 5.75tpi)
* Makes it easier to select the default feeds/speeds at startup

Sadly I'm still building the hardware so I haven't been able to _test_ those pitches on my lathe yet, but I'll be done fairly soon. ☺

johnsonm, 2023-08-27:
That took longer than I expected due to having to redesign the hardware, but all the pitches I tested so far in #255 have been correct.