clough42/electronic-leadscrew #179: ELS. Every thing connected up and looking good but motor refuses to turn. Looking for Help Please.

smbab7203, 2021-09-07:
Newbie From UK. 
Hello to you all.
 First and foremost I must say a huge thank you to James for all his hard work in the production of his youtube channel. I have been following the ELS project with great interest for quite some time now. I have taken the plunge and ordered all the parts.
I have read all the issues on this project and have learned a great deal from all your collective knowledge, so a big thank you to you all.
I have very little electronic/electrical knowledge so I am hoping you can help me with my current problem. No pun intended. 
Motor and controller is.
[image]
[image]
When I power everything up, drive looks ok no alarm LED.
The LED & KEY board looks ok, the buttons all work and when the encoder is turned by hand the read out showws the speed which I assume is correct.

When the spindle encoder is turned the motor does not turn. Thats the rub!!
I have no scope so can only check connections with a multimeter. So please forgive my lack of knowledge.

The driver is Stepperonline CL86T.  -- If I check between the following connections on the drive with my meter I get the following results when spinning the encoder by hand.

ENA - &  ENA+     On meter no change   0v.
DIR - &   DIR +     On Meter   1.3v  and 5.2v this changes depending on direction.
PUL - & PUL +      On Meter   1.3v   No change at all. Change direction no change.
ALM + & ALM -    On Meter    3.3v   

FYI. I have purchased a cheap motor frequency driver from ebay, when connected this drives the motor smoothly with no obvious problems. so I am assuming the motor and driver are not the problem?

Any help or suggestions you guys can give me as a way to resolve this problem will be much appreciated and gratfully received.

Kind Regards and Thanks in anticipation.
Steve B.

Turner75, 2021-09-07:
Oldie, originally from UK, now in Hawaii.
Sounds like no enable signal or clock pulses getting to drive. Encoder
output and buttons being processed by Ti launchPad so..... Check enable and
pulse on the interface boost board o/ps. Also the switches s3, s4 and s9
are set correctly.  The soldered connections on the boost board could be
the problem. My dim vision and shaky hand made soldering the connectors a
little tricky!
 I find that a led in series with a 4.7K resistor on flying leads can be
useful and spot many a fault that a DVM gives ambiguous readings.
 I found o/c diode on LED&KEY from Clough42 kit using the simple led tester.
 Cheers Graham

smbab7203, 2021-09-07:
Hi Graham,
Thank you for your prompt reply.
I am not exactly clear what you mean, just to be absolutly clear are you saying:- connect the LED in series with the 4.7k resistor. Then connect that in the lead running to the pin you want to check? so if the pin is carrying a signal it will light the LED?

FYI. I have purchased a spare interface board from James because I made a school boy error when soldering the LED&Key board so ordered a spare for both boards because of cost and carrage from US to UK. anyway I might just solder it up and solder a LED resistor tester as well.
Is there a specific way to solder the resistor to the LED because the LED,s are pos / neg sensitive i believe ?
I will dig out some LED,s and resistors tomorrow and have a go. I will let you know what i find.
Thanks again.
Regards  Steve.

Turner75, 2021-09-07:
HI
Sorry I was not too specific about the led/resistor test lamp
configuration. The led, being a Light Emitting DIODE is sensitive which way
round it is connected into the circuit being tested. Does not matter which
end the resistor is connected to the led but I have a  black, negative,
lead  to the led cathode, usually the lead out of the led next to the flat
on the body, and the resistor to the anode, positive, of the led with a red
lead connected to the other end of the resistor, i,e, everything wired in a
series loop.
To use, connect the black lead to the zero, 0 volt terminal on the board
and red lead  is used to poke around looking for positive volts, led will
light if the testing point finds a  positive voltage above the turn on
voltage of the diode, which is usually around 1.5 volts for a red led. If
the level is 3 volts, the led will be some what dimmer. If the test point
is pulsing, i,e, the clock or output of the encoder, then the brightness
will be less. The clock or pulse line to the stepper drive has very short
pulses which at low rev. settings may be difficult to see in normal
lighting, depending on your drive input requirements, pulsing high or
pulsing low.
Sorry the explanation is so long winded but l don't know your level of
electronic experience and I have tried to cover the very basics. Somebody
may reply that reverse connecting the led will damage it, I have not seen
that happen at the low, 5 & 3 volt levels found in these logic circuits
with a limiting resistor of 4700 ohms. This device is meant for logic
levels only, upto 5 volts max.
Cheers Graham.

smbab7203, 2021-09-08:
Hi Graham.
Thanks for another detailed reply. 
Just a little back ground info. I have recently retired after working as a service engineer on CNC Machine Tools. I was biased towards the mechanical side of things mainly. The company thought was if your happy doing electrical work fine. When you get stuck shout help and they would send an electronics engineer to help. Consiquently my knowledge of electrics/ electronics is  limited. So dont apologise for a lenghthy reply, the more information I get from you the less chance I have of messing things up. If I am unsure about anything I will Ask.
Anyway, I have sent for some resistors and leds. Please bear with me for a couple of days for delivery  and I will get on the case and let you know what I find.
Thanks again for your understanding.
Talk to you soon.

Regards   Steve.

Turner75, 2021-09-08:
HI Steve
The Github link does not support long emails. Tried sending a page or two
of background++a pic. or two, machinery of course. No success, bounced back.
Have you a regular email address?
Cheers Graham

Turner75, 2021-09-09:
> and looking good but motor refuses to turn. Looking for Help Please. (#179)
>
> HI Steve
> Seems a bit weird sending emails through clough42/ELS but......
> My background is ;-  Always tinkering with stuff, (STILL DO), from an early
> age, alarm clocks, radios, (wireless) all got deconstructed much to the
> anoyance of rest of family.. Started as electronic tech with RR Aeroengines
> in late sixties. Retired late nineties as automation production specialist.
>    Relocated to the US and worked for a small company designing and
> building realtime Xray backscatter camera systems. Upgraded a few
> Bridgeports to CNC. Retired (again) to live in Hawaii, sweet spot on planet
> earth.
> My 1924 Drummond lathe has followed me around the world since aquiring it,
> age sixteen. Started building ELS system to get rid of noisy dangerously
> exposed change wheels on said lathe which is showing its age, then decided
> to invest in new lathe, Grizzly M0602.
> Anything l can offer in the way of help will be at your request Steve.
>   Waiting for new lathe to arrive, made ELS, now up running and tested out.
> Just aquired a 2 hp motor
> Cheers Graham

---------- Forwarded message ---------

smbab7203, 2021-09-09:
Hi Graham.
How are you doing? I hope all is fine with you (In the sweet spot on planet Earth) !!
I borrowed some resistors and Leds from a mates son's science project. 
I built the tester. I am getting signals from the spindle encoder but signals going to the motordriver don't seem to change. Unless they are too quick for the LED to follow? The only one I can see change is when I spin the encoder in different direction which to me is correct?
From my limited knowledge I think the Clough interface board is playing up. I am going to solder up the new one later this evening and will give that a try. Watch this space.
I really appreciate you taking the time and trouble to help me. Thank you again!

As requested Try This    "stevebaileys@mail.com"

Regards   Steve.

smbab7203, 2021-09-09:
Hi Graham.
I have not had time to solder the new interface board yet but here are results from using the test LED.
The neg wire from LED has croc clip on it and connected to GND pin on JP7 Texas launch Board.
Using the red wire as probe these are results as best I can manage:-

From Spindle encoder to   J12.
     pin    ///////            LED On         ///////              Spinning encoder by hand Led On.

   A        /////////            no   ///////////////////////////                yes
   B        /////////            no   ////////// //////////////////               no
   I          ////// ///          yes    ///////// ///////////////////               yes
  +5v     ///////          yes   /////////// ////////////////////              yes
   GND   /////////       no   //////////////////////////////////               no

Checking J2  on Clough Interface Green connector.
 Pins//////////////// LED On.

 Plus +   ////////////    Yes
STP- /////////////    Yes
DIR- /////////////// Yes or No Depending on direction encoder is turned.
EN-  //////////////   Yes
ALM+ ///////////    Yes But Dim
ALM- ////////////     No

(Sorry about all the slash signs but it is the only way I could find to make it make sense. When going from write to preview it changed the spacing. And it did not make sense. Technology eh?)

Not sure what these results will show you ?
I will build the interface board as soon as I have time.
Talk soon. 

Regards Steve.

Turner75, 2021-09-10:
HI Steve
If I am reading this right;-
Out of encoder channel A, signals OK.
Out  of   "             "     B, signal no change.

Out of interface board, to motor drive, green screw terminals;-
STP  led is on continously
DIR.  Changes with direction
EN  led is on continuously.

Hmmmmm?
If the encoder output did not have both channel A & B pulsing, then the
system Board would not know the direction it was rotating and would not
change DIR output to the interface board. As I mentioned in last email, the
motor  clock pulses out of the interface, boost, board are very narrow and
depending on speed of encoder rotation may show up on led as no o/p, or dim
at fast encoder rotation, coarse threading, or led full on and dimming
slightly depending on settings in CONFIG.H file for pulse convention,
positive or negative going.
    Have you tried my suggestion of driving the motor drive unit with
interface board connections, green terminals, disconnected? That would tell
you whether or not the motor drive and motor are functional. Then if OK,
work back through the interface board checking input state to output state
on each line, after all, it's only function is to convert inputs from the
3.3 volt logic of the TI launch to the 5 volt logic required by the motor
drive unit and the display unit. Back in my younger days even 5 volt ttl
logic was considered too vulnerable to put in an industrial environmentso
we used HNIL, high noise immunity logic, 12 volt supply and lots of
capacitors and screening of inputs. How times have changed!
It would be clearer for me to give state of led as ON or OFF. Or, ON then
OFF.
Also l send via email or text, the logic states as _ an underline for zero
and -- hyphen for a one. OR, 0 (zero) 1 (one). Logic change states as
 Zero then upto a one, then down again   to zero ie, __--__  or, inverse,
--__--
     -_-_-_-_>, continuouse pulse train.
Keep at it, it's all a learning process.
Cheers Graham

smbab7203, 2021-09-14:
Hi Graham.
After a great deal of help and a lot of head scratching I think I have finally got the ELS working correctly and fitted on the lathe. The point of this last message is to Thank every one involved with this Git Hub project The Electronic Lead screw project.
I started watching James youTube videos and as time went on I started to get more interested and the way James explained every thing step by step, it slowly started to dawn on me I may be able to copy this and follow along and may be complete this project on my own Lathe.
To cut to the chase, I have indeed finished the ELS project and it is working.
So I would like to extend my gratefull thinks to James for all the hard work he put in in bringing this project from inception to A kit and list of parts to enable newbies like me with no electrical experience what so ever to follow along and build the project.
I must also thank Graham who spent many  hours sending me long detailed emails and instructions on how to over come my particular problems. Thank you Graham you help is much appreciated..
lastly I would like to thank you all for your different posts and questions regarding this project and all the different participents who answered these questions and posts. I have read all the posts questions and answers, you have no idea how much I have learned over the past weeks.
Thank you all.
I wish you all the best in your endevers.
Stay safe.

Kind Regards.   Steve.  UK.

closed 2021-09-14 by smbab7203

Turner75, 2021-09-15:
HI
Many thanks to everyone who has participated in this project. James Clough
has been an inspiration to us all in conceiving and implementing this
project and devoting so much time and resource to bring it to fruition.
Thank you all again.
Graham T Foulkes