clough42/electronic-leadscrew #181: Increase acceleration time?

RyanH122, 2021-09-27:
Hello. I'm trying to get the StepperOnline CL57T driver set up and thedriver blinks 7 times rapidly on the error light. 
 
The PDF says "The acceleration time setting is too small,increase the value of appropriately". 

StepperOnline says I have to do that in the controller?  How do I do that? Thanks!

Johnboy251, 2021-09-27:
You will need to download the commas software for the stepper drive to run from a pc or laptop that has RS232 comms port.
It the stepper set-up there will be a tick box to change the acceleration time.

RyanH122, 2021-09-27:
Oh geeze, thanks!  Any idea what I should set it to?

Johnboy251, 2021-09-28:
Hi Ryan.

A fast acceleration isn't necessary on the leadscrew. Buy the time the lathe motor has spun up the leadscrew will follow and will synchronise with the Chuck rotation. The leadscrew speed will still be synchronise even under load.

Back to acceleration times. I'm not in a position where I can check the Acc. Time on my system presently but anything between 2 and 5 seconds would suffice. If you can estimate how long by observing the time your lathe takes to run up from switch on that would be adequate. 

Hope that helps.

Best Regards
John 🇬🇧

RyanH122, 2021-09-28:
Hi John.  Thanks for the reply.  I appreciate the help.  I'm only trying to increase the setting as that is what SteppperOnline is telling me I need to do.  So far no luck.  Now they are saying it needs to be changed in the host software, not on the controller and I have no idea how to do any of that and it doesn't seem anyone else has to go through this.

What unit do you have?  I'm thinking this one is going back and I'm going to buy something that works out of the box.

Thanks.
Ryan

pmelevende, 2021-09-30:
Hi Ryan,

FYI, I have the CL86T driver and it works out of the box.
Thanks
Pauly

RyanH122, 2021-09-30:
> Hi Ryan,
> 
> FYI, I have the CL86T driver and it works out of the box. Thanks Pauly

Thanks!  I appreciate the information. After trying another day I'm giving up on this setup and getting something else. The best I've been able to do is to get a 1/4 turn from the motor before it errors out.

pmelevende, 2021-10-02:
No worries,

How did you go?

> > Hi Ryan,
> > FYI, I have the CL86T driver and it works out of the box. Thanks Pauly
> 
> Thanks! I appreciate the information. After trying another day I'm giving up on this setup and getting something else. The best I've been able to do is to get a 1/4 turn from the motor before it errors out.

RyanH122, 2021-10-02:
I ordered the Rattm motor from Amazon that was tested in the video. It should be here tomorrow. 😀. I'll need to make a bushing to reduce the motor pulley from the 10mm shaft I had down to 8 but that's an easy lathe project, LOL. This one should work out of the box

RyanH122, 2021-10-06:
I had to modify the motor mount plate I had made for the StepperOnline NEMA 24 motor to fit the rattm NEMA 24 motor as the bolt hole pattern is not the same but all I had to do to get the Rattm set up was set the driver box switches to 1,000 to match the motor encoder and everything is good to go!  Really liking this now!  Thanks!