clough42/electronic-leadscrew #182: A problem with the els

pikolo335, 2021-10-06:
Hi all!I had posted before and managed to make the els work out. After some time now the motor only turns on one direction...the screen was flickering before and now i move the lathe chuck on any direction and the leadscrew movee only clockwise...Any thoughts?The thing is that it was working before but not now....

SonnyMadlangbayan, 2022-07-17:
> Thanks alot i will check it!no matter how the chuck turns the leadscrew allways turns clockwise pressing the forward or reverse does nothing i will post back.Also i have not connected the ground cause i dont understand how to have a diffferenr ground...they all go to the earth right?

I have the same issue on my els. did you get a chance to fix it and what is the fix? I'm using stepper online integrated servo with no enable pin. thank you in advance.

SonnyMadlangbayan, 2022-07-22:
> I have the same issue on my els. did you get a chance to fix it and what is the fix? I'm using stepper online integrated servo with no enable pin. thank you in advance.

thought I could share this if you have not able to fix the not reversing issue on your els.

https://github.com/clough42/electronic-leadscrew/issues/154
I can't wait for the second microcontroller and I also think it will not fix the issue. so I find other solution. I notice that when putting a jump wire between -DIR and -PULSE the servo will reverse the rotation (also when spindle is forward or reverse the rotation is reverse) so I connect a wire to the spindle toggle (forward/reverse) switch connected to the -DIR and -PULSE. now when I push the toggle switch to spindle reverse direction the wire is now connecting the -DIR and -PULSE reversing the servo rotation. I test it by cutting 1.5mm thread and it is very clean.

SonnyMadlangbayan, 2022-08-23:
Please check issue#154 where I was able to fix the reverse problem. I replaced the booster board and everything works fine.

clough42, 2022-10-11:
Can you confirm, did you jumper between -DIR and -PULSE, or +DIR
and +PULSE?  I would have expected the latter, jumpering the + sides
together.

SonnyMadlangbayan, 2022-10-11:
Hi James,

yes I put a jumper switch between -DIR and -PULSE to reverse the servo rotation.
but I already replaced the booster board and now everything is working without putting jumper to -DIR and -PULSE. I suspect the first booster board Q5 is damage due when I reverse the spindle the voltage did not change.

stuwindsurf, 2022-10-27:
I also have this issue now I have installed everything. It worked fine when I bench tested it, now it's all installed the leadscrew will only turn in reverse regardless of the fwd/rev switch and if I rotate the chuck backwards and forwards by hand the leadscrew only progresses in the reverse direction? Did anyone find what is causing this other than a faulty booster board?....
Thanks Stu

clough42, 2022-10-27:
Stu,

If it worked on the bench, but only turns in one direction on the lathe, I
would suspect the connections to the -DIR or +DIR terminals on the servo or
stepper driver.  Both +PUL and +DIR must be connected to the + terminal of
the booster board, and -DIR must be connected between the boot board and
the driver.

If you're using an integrated servo, I did just upload a new diagram
showing the connections:

https://github.com/clough42/electronic-leadscrew/wiki/Wiring-Everything-Together

James