clough42/electronic-leadscrew #184: Grizzly G0750G Implementation

K5FR, 2021-10-19:
I just finished installing the ELS on my G0750G lathe and am totally happy with the result. My biggest concern when starting this mod was finding a stepper/servo that would handle the load of this lathe. I settled on a ClearPath CPM-SDSK-3446P-RLN servo which is probably an over-design, but I didn't want to have to try the lesser motors and spend a lot of time swapping them out for bigger ones. Mounting the motor was about the same as James did his using the gear spider with the exception I used a 5mm HTD belt and pulleys. Everything else was just follow the instructions James provided. I also added a stop switch for turning to a shoulder and it works well.

pmelevende, 2021-10-20:
Yes the videos, instructions, and answers to peoples questions make this project a standout. I also completed mine recently and couldn't be happier.
Thanks
Pauly

koenbro, 2024-06-25:
> I just finished installing the ELS on my G0750G lathe and am totally happy with the result. My biggest concern when starting this mod was finding a stepper/servo that would handle the load of this [...]

Can you please tell me how you wired the Clearpath servo? Thank you kindly.

K5FR, 2024-06-25:
See this entry in my Toys blog for how I did the basic installation.
https://www.k5fr.com/Blog/?p=338

koenbro, 2024-06-25:
Very helpful thank you. I managed to get it turning.