clough42/electronic-leadscrew #187: Can't initialize els-f280049c in Code Composer

AK-Mike, 2021-11-11:
When trying to initialize the els-f280049c in Code Composer, we get warnings & can go no further. We're complete novices are trying to follow clough-42 step-by-step. Please be gentle with us.

10:16:43 **** Clean-only build of configuration Debug for project els-f280049c ****
make clean 
Cannot run program "make": Launching failed

Error: Program "make" not found in PATH
PATH=[C:/ti/ccs1100/ccs/eclipse/jre/bin/client;C:/ti/ccs1100/ccs/eclipse/jre/bin;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\WINDOWS\System32\OpenSSH\;C:\Users\rudin\AppData\Local\Microsoft\WindowsApps;;C:\ti\ccs1100\ccs\eclipse;]

10:16:43 Build Failed. 1 errors, 0 warnings. (took 469ms)

**** Build of configuration Debug for project els-f280049c ****

Nothing to build for project els-f280049c

w180m110, 2021-11-11:
I’m getting the same thing and have determined the cause to be CCS version being too new.  Let me guess, you installed the latest version available, right?  So did I, but that seems to be the problem.  I’ve managed to update the project to use the new compiler but it then throws dependency errors regarding header files.  While some were easy to fix by adding the required “includes” directory to the settings, it still bombs later on other header files.  That’s where I stopped last night, and today I was going to give up and install a previous version of CCS that is known to work with this project.

AK-Mike, 2021-11-12:
Yes...the latest version 11.0.0.0012

Please let me know if it works & please send a link to a version that works for you.

AK Wintzer

w180m110, 2021-11-12:
Here's the formula that worked for me:

1.  Download and install the latest of the CSS10 series, which for Windows is 10.4.0.00006

2.  After initial start up, wait a short while until the message pops up stating that an update is available.  Click the message and apply the update.  A restart is required.

3.  In the clough42/electronic-leadscrew git repo, click on branches (there are 6 of them at the time of this writing).  Look for the the branch that is an updated project for CCS10.  Download and use that branch.

When I did all of the above I managed to compile without errors.

AK-Mike, 2021-11-20:
> Here's the formula that worked for me:
>
> Download and install the latest of the CSS10 series, which for Windows is 10.4.0.00006
>
> After initial start up, wait a short while until the message pops up stating that an update is available. Click the message and apply the update. A restart is required.
>
> In the clough42/electronic-leadscrew git repo, click on branches (there are 6 of them at the time of this writing). Look for the the branch that is an updated project for CCS10. Download and use that branch.
>
> When I did all of the above I managed to compile without errors.

Thank you for your help. We were able to do steps 1 & 2. We found the branch showing update for CCS10, but still had warning messages in the build.

We decided to go into the ‘Way Back Machine’ by following the clough 42 video step-by-step, using CCS v9.1.0.00004 & the 1st release of Clough’s ELS program. We were able to flash the program onto the TI board, but could not get the Clough 42 readout & controller to function. The readout is getting power, but there are no numbers illuminated.

We’ve ordered a new ‘LED&KEY’ board. Hopefully we’ll at the very least, get the oldest version Clough’s ELS running.

AK Wintzer