clough42/electronic-leadscrew #190: Possible to add a super fine finishing speed?

RyanH122, 2021-12-06:
Not sure if this is the right place to ask but I'd love to see a super fine finishing feed added to the code. The .001" can be a bit too course for a couple parts I make and I'd like to see .0003" or even better .00025" added. The display could be something like SF for super fine or even just FINISH. Is something like this even possible?  Thanks!

pmelevende, 2021-12-06:
I like that idea - I have found the same and it would be good to have Sf, SSF and XSF :-)

RyanH122, 2021-12-07:
> I like that idea - I have found the same and it would be good to have Sf, SSF and XSF :-)

Exactly right!  I just know nothing about code, coding, or even if there's "room" in the code for a few more options. Maybe James or someone who does know will comment as to if this is possible.

pmelevende, 2021-12-08:
It's code - Anything is possible. I have refrained from touching it though, as any of my mods would then need to be implemented over future 'official' releases. I'm so looking forward to the next version :-)

tomvikse, 2021-12-28:
In Tables.cpp

Add this line

 { .display = {BLANK, ZERO | POINT,       ZERO, ONE}, .leds = LED_FEED | LED_MM, HMM_FRACTION_FEED(1) },

So it locks like this from line 184

const FEED_THREAD metric_feed_table[] =
{
**{ .display = {BLANK, ZERO | POINT,       ZERO, ONE}, .leds = LED_FEED | LED_MM, HMM_FRACTION_FEED(1) },** 
 { .display = {BLANK, POINT,       ZERO,  TWO},   .leds = LED_FEED | LED_MM, HMM_FRACTION_FEED(2) },
 { .display = {BLANK, POINT,       ZERO,  FIVE},  .leds = LED_FEED | LED_MM, HMM_FRACTION_FEED(5) },
 { .display = {BLANK, POINT,       ZERO,  SEVEN}, .leds = LED_FEED | LED_MM, HMM_FRACTION_FEED(7) },
{

Now you get 0,01mm (metric) feed that is .00039 feed