clough42/electronic-leadscrew #194: TI Compatible board

Flyingtux, 2021-12-19:
Hi,
Could you list or references all TI compatible board somewhere to increase the choice. It's too hard to find a XL-F280049C right now. Thanks in advance

abelshe, 2021-12-19:
I’m interested in this as well, following! I haven’t been able to find a single US distributor stocking the TI LaunchXL-F280049C.

clough42, 2021-12-20:
This issue appears to be a shortage of the F280049C microcontroller chips
themselves, and not just the LaunchPad boards.  I see zero in stock
anywhere, globally, including the Chinese PCA assemblers.  Without chips,
TI can't assemble more LaunchPads.  We've been very fortunate with this
project to have dodged the effects of the global chip shortage--until now.

Porting the code to the F28069M board would be possible, but not without
changing the hardware.  TI doesn't honor their own standard pinouts from
board to board and chip to chip so things like the SPI clock pins are in
different locations, and also the power domains are different, requiring a
different power supply and voltage regulator configuration.

I don't think it makes sense to re-spin the hardware and software to try to
adapt at this point.  I see fewer than 100 assembled F28069M boards
available globally at the moment, and zero additional chips in stock
anywhere.  I think it's very likely the F28069M boards would also be out of
stock and unavailable by the time I could re-spin the hardware, modify the
software, and get it tested and ready for release.

Another possibility would be a port to an entirely different MCU platform,
but that doesn't do much good if those chips are out of stock, too, or if
the TI chips are back by the time that effort is complete.

James

jalfeld, 2021-12-21:
I was just able to order one directly from TI, they are showing them as in stock....

clough42, 2021-12-21:
Oh, awesome!

tpconst, 2021-12-31:
I live in Canada, just ordered and received a board from DigiKey.ca (roughly 40 in stock). Don't know if they will ship to the US

PaulCElliott, 2022-01-01:
I've Just ordered an F280049C from digikey in the UK and its been shipped. Just followed the link from the youtube series.
I guess the work now begins :)

JonWoellhaf, 2022-01-01:
I would say the FUN now begins!

From: PaulCElliott 
Sent: January 1, 2022 13:17
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] TI Compatible board (Issue #194)

I've Just ordered an F280049C from digikey in the UK and its been shipped. Just followed the link from the youtube series.
I guess the work now begins :)

PaulCElliott, 2022-01-02:
You may well be correct:)

For interest my target is a WM250V Warco lathe, so not very different in 
size to the one in the clough42 videos. I'm planning to breadboard it 
with two PICs simulating the encoder and stepper. I've got the silly 
idea of trying to come up with an alternative to using the halfnuts as I 
sometimes seem to engage them in a position fractionally before the 
number I'm aiming at. I wouldn't want to debug that on real hardware.

All the best,

Paul

Flyingtux, 2022-01-22:
Hi,
Thanks for sharing,
I bought and receive mine from aliexpress, the board price was bigger (55€) than on TI website but shipment cost to France was much more affordable 5€ (Taxes includes), so 60€ at all.

Cheers,

bdlabitt, 2022-02-13:
Mid-February 2022, the chip/board drought continues.  Last fall, I bought a used G0752Z lathe, which is basically a G0602.  Been looking for a suitable Launchpad, but so far, none to be found.  Puts this cool project on hold for me, until a board can be sourced.  Threading on a 602 isn't hard, but it is a bit messy and cumbersome with the clips.  Would like to upgrade to an ELS.  Availability of boards is just a guess, one distributor posted 7/29/2022.  Going to be a long wait.

tekym, 2022-04-27:
End of April 2022 update: Still no XL-F280049C availability anywhere.  Mouser claims to expect shipments at the end of July and September, and then at the end of May 2023.  I don't know how trustworthy those claims are, or whether the few hundred they're expecting to receive will sell out immediately or not.

I don't know the first thing about porting software between platforms, but I do note that the Teensy 4.0 and 4.1 (ARM Cortex M7) have quadrature encoder inputs, floating point support, and presumably enough general power to match the TI boards as James noted in one of the early videos.  They're also in stock at distributors and continually refreshed, [it sounds like](https://www.pjrc.com/store/teensy40.html).

cstoker2, 2022-05-01:
I agree that a teensy 4.0 should work well, it has libraries for quad encoders: https://github.com/mjs513/Teensy-4.x-Quad-Encoder-Library
for fast stepping (300k/s):  https://github.com/luni64/TeensyStep4
and for the tm1638: https://github.com/gavinlyonsrepo/TM1638plus

clough42, 2022-05-03:
The news I'm hearing indicates that the TI boards should be available again
in a couple of months and things should start to ease beyond that.  I
cannot see the future any more clearly than anyone else, though.

I think the Teensy 4.0 is an attractive option for this type of project,
and is something I'm looking at for a future project.  One downside of the
Tesnsy is that the hardware quadrature decoder has a much shorter register
than the TI DSP chips, so it rolls over much faster.  This should be
manageable, but I haven't had a chance to play with it yet.

James

PaulCElliott, 2022-05-04:
I must have been lucky to get mine when I did. I'll order a spare when the supply situation eases.I've got it all running on a WM250 V Warco lathe Incidentally; it works fine. Thanks for all the effort developing this.Is there anywhere we can list the machines that its been installed on to help others in future?Regards,PaulFrom Paul ***@***.***  ---- On Tue, 03 May 2022 15:40:51 +0000  ***@***.***  wrote ----

The news I'm hearing indicates that the TI boards should be available again

in a couple of months and things should start to ease beyond that.  I

cannot see the future any more clearly than anyone else, though.

I think the Teensy 4.0 is an attractive option for this type of project,

and is something I'm looking at for a future project.  One downside of the

Tesnsy is that the hardware quadrature decoder has a much shorter register

than the TI DSP chips, so it rolls over much faster.  This should be

manageable, but I haven't had a chance to play with it yet.

James

bdlabitt, 2022-05-05:
From looking this up on a Teensy 4.1, the position register is 32 bits.  It seems the TI part is the same.  Is there a different register that is smaller than 32 bits?

jimhinze, 2022-05-06:
FYI everyone... DigiKey has 55 in stock (I just bought 2)

https://www.digikey.com/en/product-highlight/t/texas-instruments/launchxl-f28069m-c2000-piccolo-launchpad?utm_adgroup=Texas%20Instruments&utm_source=google&utm_medium=cpc&utm_campaign=Dynamic%20Search_EN_Focus%20Suppliers&utm_term=&utm_content=Texas%20Instruments&gclid=CjwKCAjwjtOTBhAvEiwASG4bCNziRyS9lcfhHneGsZPYLKvzNXWoXjR9Q70MtTmliG9C21z1EXmcuxoCU1EQAvD_BwE

tekym, 2022-05-06:
Check James' comment in this thread, the 3rd post.  The 28069M isn't the same board, it's got different pinouts etc.

tekym, 2022-07-01:
Status update: I just got an email a couple days from Mouser saying "Production Completed JUN 27, 2022 - In transit to our Warehouse", with an estimated ship date for my backorder of July 26th, so a couple hundred of these TI boards are now on the slow boat from Taiwan, probably.

digiexchris, 2023-01-27:
Bumping this again because it seems we're globally out of stock again. I can only buy one on ebay for $359!!

Any additional thoughts on adapting to the teensy?

clough42, 2023-01-28:
I see more than a thousand in stock at Mouser.

OG1-SS, 2023-01-28:
There is one on eBay now at a good price but shipping is high.

bdlabitt, 2023-01-28:
Yes, there are loads of them still at Mouser, but zero at Digikey (yesterday).  Due to the shortages last year, I designed my own ELS using a Teensy 4.1 and a touch panel display.  Runs great.  Wished I didn't have to do that, because I think the ELS on this site is quite good.  But from Nov 2021 to May 2022, there weren't any LaunchXL'-F280049C's to purchase.  One month after I got mine working, the LaunchXL's were back in distribution!  But I did learn to design my own PCBs, so it was time well spent.  In my opinion, a touch panel display is a superior user interface, but buttons sure are a lot easier to code!

bdlabitt, 2023-01-28:
Mouser website has the boards for $51.87 USD.  Their shipping is ok.