clough42/electronic-leadscrew #195: Stepperonline iSV57T and KL-5080H

jscottmoss, 2022-01-01:
So I had to put this project on hold while we moved and a year later I am finally getting back to this thing. I had to get a new motor b/c the other one was lost never found when we unpacked. I bought the motor/servo whatever James showed on youtube from Stepperonline. I am trying to wire everything up, and I've hit a snag again. 
not sure where W V U wires should go to the motor from the KL-5080H or is that not needed with the Stepperonline iSV57T ? 

Thanks

tomvikse, 2022-01-02:
The servo from stepperonline have the driver onboard,so no KL-5080H is needed.

Clausing5902, 2022-10-02:
Should an updated diagram be made to address the difference in wiring without the KL-5080H....or is it all that different?

puneeth-kvsk, 2023-01-06:
So might one communicate with ISV57T using a custom program (ex. arduino , etc) ? Can somebody please share their valuable experience regarding the same?