clough42/electronic-leadscrew #196: No display on the Control Panel

Howard-001, 2022-01-05:
All I'm seeing on the led/button panel is the power indicator.

I've got two launchpad boards and I've tried both 1.3 & 1.4 software on each, both seem to be loading correctly in either debug or flash.

I've tries two different cables to the control board & buzzed them both to verify they're good.

I have continuity from the input connect to the appropriate pins on the display board driver.

I've checked the inputs to the adapter board, I see clocking on MOSI , constant high on DISP and open collector on CLK.

I've measured the input resistances on the adapter card, they measure ~ 560 ohms as I would expect from the schematic.

With the adapter removed from the launchpad & powering it via usb I see clocking on MOSI (J6_55) but open collector on DISP (J6_53) and CLK (J8_71).

I've doubled checked the settings foe S3 a/b, S4 & S9 and flipped them back & forth to ensure good contact.

Anyone got any ideas?  I hate to think I got two defective TI boards.

pmelevende, 2022-01-08:
I Howard,
I'm just replying as nobody else has yet - Do you have a spare display board to try? That'd be my next port of call.

Thanks
Pauly

Howard-001, 2022-01-08:
Thanks for the response Pauly, digging deeper I've found that the Q2 circuit on the interface seems to be the Root of all Evil, the +5 V that's supposed to be on the supply side of R4 wasn't there; I touched up the solder connections & I now have a high on the interface clock pins , however when plugged into either launchpad  I'm still not seeing any clock pulses, it just sits at high.

I've just ordered another interface & display as well as another launchpad , hopefully this will produce a working unit.

WheatMillington, 2022-01-13:
My experience is that the display boards are fragile and have poor QC, so hopefully that will resolve your issue.

WheatMillington, 2022-02-09:
I also managed to kill a board with sloppy desoldering.... ok two boards. My preferred solution is actually to just clip the pins very short and solder to them on the topside. You need to remove the little yellow plastic header thing, which is easy.

Howard-001, 2022-12-03:
Finally got back to this, after verifying the interce board9s) on a test bed and the display9s) with an arduino I tried a new installation of CC on a fresh machine with a fresh OS and a fresh launchpad & I've now got things displaying :)

My two other launchpad  cards still won't work :(

closed 2022-12-03 by Howard-001

Chucksbp, 2022-12-03:
Sounds as though the electronics industry is also having quality issues!!

Howard-001, 2022-12-04:
> <https://github.com/clough42/electronic-leadscrew/issues/196#issuecomment-1336193013>view
> it on GitHub, or

From the number of others that seem to have had 
the same run of "bad luck" I'm suspecting the 
boards could benefit from some static suppression 
measures to protect them from people with 
electric personalities such as myself :)

Chucksbp, 2022-12-04:
Yes That's very possible. I'm sure that as complete electronic novices as we are, proper precautions are not taken.  I had a number 1 encoder input failure and a display board that never worked. I did after the display failure get a static pad for my bench with a wrist band but it may have been too late for the board input. Although I blame that on  the encoder????? 
Luckily since the initial problems everything has been working for close to a year now, much to my enjoyment!!! the lead screw is a great addition to my old Atlas lathe! I do have a complete unit as a spare except for a motor. the board id programed and tested and ready to go!
As always thanks to James for all the effort to build and share such a wonderful piece of equipment.

Howard-001, 2022-12-04:
As embarrassing as it is I've been in electronics for 50 years :(

It's just something about me- I've got zapped 
more times than I count walking into a server 
room where everything is grounded & just touching a relay rack :O

God I miss vacuum tubes.