clough42/electronic-leadscrew #197: Need help troubleshooting, no RPM shown

RyanH122, 2022-01-13:
My ELS has been great until today. For some reason it quit displaying / seeing spindle RPM. I checked all my connections and everything is good but I replugged them anyway and no change. I got my volt meter and checked the encoder connection at the TI board and there's 4.98v at the TI board so I'd say it's supplying power?  I also check the other pins on the TI board with the encoder connected using the ground on the board and got 4.98 at 5v, 4.96 at 1I, 4.83 at 1B and 1.41 at 1A. 

I get the same readings if the lathe is running or not and they don't change if the lathe is running forward or in reverse. I don't know what I should be seeing or what else to check. 

Any help would be greatly appreciated. I was just about to start a cross slide screw extension. 

Thanks!

Turner75, 2022-01-13:
HI Ryan?
If you are not seeing th A - B outputs from the encoder change state when
rotating slowly by hand and + 5volts and 0 volts are present on the encoder
lines, then looks like the encoder is toast. Check if there are the A - B
complements available from the encoder, usually marked up A, or B with a
bar over the characters.
  Or, you could try switching the encoder input to the second row of
connectors for encoder tracking inputs and change the designation on the
software over in the configuration H file. Good luck.

RyanH122, 2022-01-13:
Hi Turner.  With the encoder connected to the TI board the voltage readings don't change. They're the same with or without the encoder turning in either direction. 

Would it be better to power the encoder separate, say off a small power supply and then check the readings?  I honestly didn't think of that earlier. 

Thanks!  Ryan

pmelevende, 2022-01-13:
Hi Ryan,

Is the stepper still turning the lead screw correctly? If so, the encoder is working fine.
Thanks
Pauly

>

RyanH122, 2022-01-13:
Hi Paul. No, the stepper isn't running either.

pmelevende, 2022-01-14:
Hi again Ryan,

In that case, I'd suspect either the encoder or TI board interface. As you have checked the encoder output, and its not changing, I'd suspect it to be the culprit.

Thanks
Pauly
>

Chucksbp, 2022-01-14:
Ryan, I am having the same issue. my ELS was working fine and just stopped. Its a new build actually just had it up and operating a few days. I'm trying an encoder first.  I did the same tests as You.

RyanH122, 2022-01-14:
Hi Chuck. That’s what I’m doing too. I’ve noticed these encoders cost anywhere from 20ish bucks up to pushing 3 hundred. I wonder if the lower cost ones are inferior quality?  

Mine is / was about 5 months old and gets weekly use but not a ton. I have my fingers crossed it's the encoder and not the TI board as those don't seem to be available anywhere. If you get yours soon post back if it fixes the issue and I'll be sure to also update.

Chucksbp, 2022-01-14:
Ryan, I am sure these are Chinese knock offs. My installation is on an Atlas 618, so the encoder has never seen 1,000 RPM.
When mine quit I was just playing with the feed rate on the controller when the RPM went away and the motor then stopped. my lathe is strictly hobby, im a retired gearhead 72 years old, soldering the clough42 board was a real challenge with old shay hands. if anything i was afraid of the initial build having issues not after it was up and operating as it should..
I really hope its the encoder only for the sake if it is the board, what actually caused the problem. 
All my buttons work as they should on the control panel. when on the motor is locked and when the power off button is pushed the motor is free to spin. so in my mind most of the chip and the interface is working. 
I would be afraid to spend big money on a supposedly Genuine encoder for the fear you would be getting the same cheap one for big money??? 
I will keep in touch if you do Facebook feel free to use Messenger if you would like to communicate, My name is Charles Mayerchak.

Turner75, 2022-01-14:
HI Ryan
Twenty dollars won't buy much encoder. A set of quality bearings for a
device that has to spin long and fast might cost that. I was fortunate in
having a couple of Swiss Baumer encoders, 500 lines, which came off a
machine  that had thousands of hours of high speed operation and is still
going strong.
  Check around the encoder for any oil/fluids present, depending on how
well they are sealed, contamination will kill them. Also, they have to be
mounted to minimize side loading of the shaft. Too much side loading will
quickly result in bearing trouble with cheap units and that will kill them.
Check the encoder shaft for wobble. Any side play could cause the encoder
disc inside the unit to rub against the opto assembly and score the pattern
off the disc. Bottom line:-

1. Remove encoder from lathe.
2. Check for contamination
3. Check for end play on shaft
4. Supply encoder with 5volts from a reliable source.
5. Using either an oscilloscope, digital volt meter, (DVM)  or logic probe,
( or, possessing neither, an LED with a 1000 ohm resistor in series to
limit the current will suffice),  check for change of state on A B output
lines while turning encoder by hand slowly.
Note* some encoder outputs require a load resistor  connected to 5 volts as
the outputs are open collector transistors. Check with data sheet for the
device under test.
Good luck.
Cheers Graham

RyanH122, 2022-01-14:
> Ryan, I am sure these are Chinese knock offs. My installation is on an Atlas 618, so the encoder has never seen 1,000 RPM.
> When mine quit I was just playing with the feed rate on the controller when [...]

Mine did the exact same thing, the only difference was I had ELS off so the motor wasn't turning as I didn't need it on the project I was doing. The display just went to 0 on the RPM. Like yours, my buttons all work, etc.

RyanH122, 2022-01-14:
> HI Ryan Twenty dollars won't buy much encoder. A set of quality bearings for a device that has to spin long and fast might cost that. I was fortunate in having a couple of Swiss Baumer encoders, 500 [...]

The encoder seems physically fine and I agree that for 20/25 bucks you're not getting the best one going. If I had the 500 and change that one costs at DigiKey I'd definitely buy one but I'm hesitant to even spend 50 or 60 on one not knowing if it's any better than a 20 dollar one. Do you know of a way to tell a difference in them?

I will try and do some more testing on the encoder  but I'm really betting that it is the problem now hearing about Chuck's doing the exact same thing mine did.

JonWoellhaf, 2022-01-14:
Hi, Guys.

I had the thought that the internal mechanism of the cheapest rotary shaft encoder is likely fine. It’s the housing and bearings that are likely cheap junk.

So how about making our own housing and using good bearings? We have lathes, and it should be a fun project.

Yan

Chucksbp, 2022-01-14:
Just a hunch, but the more expensive encoders seem to have a yellow or orange stripe on the label of the encoder?? James had the link to eBay as far as purchase, although it does include some higher priced ones. I am really hopping its the encoder and not the Launch Board!

RyanH122, 2022-01-16:
Just to verify, there should be voltage, approximately 5, present on the three phase pins and V+ on the TI board without the encoder connected?

Chucksbp, 2022-01-16:
That's good to know. Did you happen to get yours operating yet??? I should have a new encoder tomorrow from Amazon if delivery is on time. I'm hoping it Fixes the problem. how long did you have your encoder and do you remember the sellers name? mine came from eBay the seller was Haititools1 I ordered it Dec.24  2021
If not Ill check those voltages. with the encoder connected i did have the *5 volts on the line pin and i had i believe 4.7 volts on A and B  I dont think I had that on The Z though. but as I said that was connected.

RyanH122, 2022-01-16:
> That's good to know. Did you happen to get yours operating yet??? I should have a new encoder tomorrow from Amazon if delivery is on time. I'm hoping it Fixes the problem. how long did you have your [...]

Sorry, Maybe it didn't come across right. I'd like to know for sure what or if the TI board is supposed to be outputting 5 volts on all the pins except the ground pin without the encoder connected. I actually have two different voltage outputs depending on which set of pins I check. The default set, the ones in the middle of the board, are 4.98 (+5 pin, 1I, 1B) and .54v on 1A. If I check the other set towards the edge of the board I get 4.98 volts on all the pins. So 1A and 2A are very different from each other. I have no idea which one is right, or if either one is right. 

No, mine isn't working. I got a new encoder today from Amazon and tried it as soon as I got to the house after work. No change. 

My original encoder came from Amazon about 5 months ago and after chatting with them they are going to refund me for it.

Chucksbp, 2022-01-16:
You have to go into the configuration H file and change the pin designation and re flash the board  for the second set of pins to work. If that doesn't fix the issue i would say it has to be the Launch board. its line 103 in the configuration H I believe. You would disable line 102 and enable line 103. I didn't want to mess with the board until I  tried the encoder. I did Take the belt off the encoder today and i see nothing mechanically wrong with it. James had an issue with a board but it was drawing excessive current and the chip was getting hot, but I don't notice any overheating at the chip.

RyanH122, 2022-01-16:
> You have to go into the configuration H file and change the pin designation and re flash the board for the second set of pins to work. If that doesn't fix the issue i would say it has to be the [...]

I was literally just about to update... I did fight my way through flashing the TI board to use the second set of pins and it's working

Chucksbp, 2022-01-16:
I absolutely agree, I wouldn't take a chance on re flashing nor would i take a chance of using the old encoder. the old encoder may have ruined a resistor or something on the board and may ruin the second set of terminals. Was i correct with the change in configuration H??? did you just change the one setting and then re flash it??

RyanH122, 2022-01-16:
Yeah, there's the one line that has to have the // added to comment it out and then remove the // from the line below to enable the second set of pins in configuration.h

I actually uninstalled code composer, deleted the workspace file and firmware and redownload the firmware, and reinstalled code composer to try and do it like a fresh new board I had to set it up for my lead screw and 3:1 drive ratio again but I wanted to software as "fresh" as possible and I couldn't find anything on how to do whatever the equivalent of a format would be on the TI board.

Chucksbp, 2022-01-16:
I have installed "Tunes" on trucks I have owned. I use a stand alone device to install the Tune but its basically the same thing but takes much much longer to clear and write the tune to the chip. 
I'm glad you were able to get it up and operating, I hope mine goes as well. I did find a board and ordered one I don't know when it will arrive. Its coming from another country by way of DHL the first one I got came from I believe Japan and took 8 days. I ordered mine Late Thursday evening. I defiantly wont hurt to have a spare as they are almost impossible to find right now. All the large suppliers are saying June+ of 2022

RyanH122, 2022-01-16:
Keep us posted!  Hopefully you can get up and running soon!  I didn't realize how hard it was to find the ti boards right now but it seems the chip shortage is hitting everything.

tomvikse, 2022-01-16:
How did you mount the encoder?
I use 2 bearings so i get no load on the encoder.
( Fidgetspinner bearings)
[image]

Chucksbp, 2022-01-16:
My encoder is driven from my spindle with a GT2 5mm belt using two 60 tooth pulleys.

RyanH122, 2022-01-16:
7x14 lathe here. My encoder is also belt drive. The bracket is using one of the original gear cover mounting holes and the pulley on the spindle shaft replaced the original thread train driving gear. With both pulleys being 60 tooth and not requiring any real power to drive the encoder it requires practically no belt tension. 
[image]

And to add... While the ELS is back to working it's not correct yet. For some reason the feed is way too fast. The encoder I bought is a different brand but still a 1024p/r but I'm wondering if it's a quadrature encoder. Feed seems like it could be 4x too fast. I had to quit last night so I'll try more tonight after work.

Chucksbp, 2022-01-16:
That's a very nice installation, very neat and tidy!
When you say your feed is too fast, are you meaning your RPM reading is too fast or your Lead screw is too fast?? If your RPM seems correct then you need to change things in the motor count. If the RPMs are fast I cant answer that, you may have to talk to James about that. 
When mine was working everything was right on. My RPMS were right on and the lead screw was on. I actually cut a couple threads as a trial. 
  I did notice that i was slightly off at first. Believe it or not all 60 tooth sprockets are not created equal. I had two different brands and one was slightly smaller diameter although 60 tooth and this would actually slowly move the two out of time. not by much but after a half hour of operation it would be close to a turn off??????? drove me crazy for a while. Even same brands will be .005 difference in diameter. I am within .001 right now with the two i am using. I felt 60 tooth is 60 tooth but there must be enough tooth to belt slop to allow movement????

tomvikse, 2022-01-16:
my setup
[image]

Chucksbp, 2022-01-16:
That's a great set up for the encoder, you have a lot of room to work with there. I am very cramped for space on a 6 inch lathe plus i didn't want to change the original appearance. I wanted to keep the encoder drive under the original gear cover without cutting the cast iron cover.

RyanH122, 2022-01-16:
Thanks Chuck. My goal was to keep it simple as I had to do it with hand tools so lots of hacksaw and filing, LOL!

[image]
[image]
[image]
[image]
[image]

That's the stepper motor mount which mounts by the single smaller hole where the banjo used to be, allowing the whole mount to pivot to adjust belt tension. It works great and wasn't too bad to build with hand tools. 

On the speed, thinking about it, the encoder has to be a quadrature as the displayed spindle speed looked about right. The lead screw is turning way too fast though. I'm not sure what's off. The encoder is set to 4096 and I have the micro steps set to 3 and 1000 motor steps as I have a 3:1 drive reduction on the motor / lead screw. And the TPI on the lead screw is 16 which is what it is. Those are the only three things I remember having to set the first time around so I'm not sure what's going on.

Chucksbp, 2022-01-16:
That's what my settings are. did you check your motor dip switches to be sure they are set for 1,000??? my motor was not set for 1,000 when i got it. there should be a chart on your controller.

RyanH122, 2022-01-16:
> That's what my settings are. did you check your motor dip switches to be sure they are set for 1,000??? my motor was not set for 1,000 when i got it. there should be a chart on your controller.

That's part of what has me so puzzled. Out of the box I didn't have to touch the stepper driver. I hooked everything up, went into the software and set the numbers, flashed the TI board and was done. It worked perfectly. I do a fair amount of threading and if I don't make a mistake the threads have always been right on. Now with this new encoder it's not close, LOL. I'm getting more and more tempted to try the old encoder this evening

Chucksbp, 2022-01-16:
In your set up file is the encoder set up for 4096 encoder resolution? check your encoder it should give the pulse resolution it should have 1024 P/R on it. 
  I JUST received the one from Amazon and i am totally disappointed!!!  Its in a brown box with Berm as the trade name. The box was opened although the plastic inside is not. there is a label over the specs on the box. looking at the unit through the plastic bag i see a 300P/R on it. I don't think I am going to touch it. I think its going back!!!

Chucksbp, 2022-01-16:
I'm going to have to give that encoder some thought. on the instructions which is mostly Chinese it does state 4096 pulses per revolution. Its not going to happen today. i am in south Western PA and the snow has started. Its almost dinner time and i'm not in the mood to fight this tonight. Let me know what you figure out though.

RyanH122, 2022-01-16:
I'm going to double check my config.h file when I get to the house after work in case I entered the encoder resolution wrong like 4069. It does say 1024p/r on it. This is the one I ordered this time. I specifically didn't get a omcron one this time as it seems real ones? are crazy expensive from DigiKey, etc so these 20 to 30 dollar ones... I'm thinking they're fake?  But I sure don't know. 

[Here](E6B2-CWZ6C Incremental Encoder Quick Response Incremental Rotary Encoder Internal ASIC Devices DC5~26V OD φ38mm(1024P/R) https://smile.amazon.com/dp/B08RS6M32J/ref=cm_sw_r_cp_api_glt_i_CBABS66GX5G4QGSXCSXF?_encoding=UTF8&psc=1)

Chucksbp, 2022-01-16:
Is that link the one you got for a replacement or is it the original one??

RyanH122, 2022-01-16:
That's the replacement that I got yesterday.

RyanH122, 2022-01-17:
Chuck, I think I have it all sorted out now.  I was going to remove the TI board and double check all the software numbers but first I checked the dip switches on the driver per your suggestion.  I didn't have to touch them the first time.  I checked the chart on the driver and set the switches for a 1,000 step motor.  The speed and sound of the stepper motor sounded much more "normal" so I set the ELS to .001" feed and put a digital caliper between the head stock and saddle, engaged the half nuts and turned the chuck by hand to feed the saddle in.  Each revolution of the chuck moved the saddle .001" just like it should so I then repeated at .010" feed and .040" feed and the saddle moved exactly what it should.  I backed out and set it to .002" feed and rotated the chuck 10 times and it moved .020" so I followed that up by cutting a M8 x 1 thread because I cut that a lot so I know how it should come out and it came out right on.  I have no idea why it worked fine with the default switch settings before or why I had to change them for this encoder but I'm just glad it seems to be working properly again.  :-D  I plan on doing some more hand feed tests and a couple more threading operations tomorrow.

On the encoder I got, it is a BERM.  I was pretty sure but verified once I got to the house.  The box is a fairly professional looking, dark brown on the front going about 1/3 of the way around the sides and there's a lighter diagonal stripe on the front.  There is a specs sticker on the one side of the box stating BERM® and under that Rotary Encoder B E6B2-CWZ6C 1024P/R DC5-24V(Vcc)  One thing I noticed its that this encoder seems more precise(?) than the OMRON one I was using.  This one, I have to set the belt tension tighter as if it's as loose as I had it before I can hear the stepper motor vary RPM as the belt flops a bit changing the encoder RPM slightly and if I turn the encoder a tiny tiny bit, just in the looseness of the belt the stepper responds.  That didn't happen with the Omron one.  I could turn the encoder pulley in the slack and the stepper didn't respond at all.  So maybe this is a slightly more accurate encoder?  I don't know but if it lasts I think it's going to be okay.  :)

Chucksbp, 2022-01-17:
That's good news!! That's the same brand encoder i got, but the one I have is 300 P/R rather than 1024. I can change that in the H configuration. I don't know how much that will affect the accuracy but maybe ill give it a try. As i said Ill have to mess with a bunch of snow first. so I don't know if it will happen tomorrow or Tuesday. I'm 72 years old so the snow may wear me out! LOL

RyanH122, 2022-01-17:
Be safe in that snow!  I'm in NE Ohio and there's 6+ on the ground already. It has drifted up a foot or more on the side of my car. 

A 300p/r encoder is still 1200 pulses per revolution right?  That's about a 1/3 of a degree per pulse?  I got a 1024 because I was hoping to just plug it in and be set. I'd like to know more about how that all works but I'm fairly lost when it comes to the software side of all this. That's the great thing about Amazon, they have a fantastic return policy if it doesn't work the way you want it to if you decide to try it. :-D

Chucksbp, 2022-01-17:
Well I got my snow duty finished! We had around 8 inches or so.  i doubt that I will do anything with the ELS today. I didn't realize that I wasn't getting a 1200 pulse encoder, it was my fault for not reading everything and being in a hurry. I will probably order the correct one and possibly wait it out??? I also got conformation on the Launch board. It will be here Friday the 21st Via DHL. its coming from the UK. I was a little nieve when I started this project a month ago. I sure learned a lot doing it. 
What is your actual thought was it the Board primary connections or the encoder??? 

  I am still trying to find a way to wire a carriage stop into the ELS. not for threading just an emergency stop for regular turning. just so you cant crash into the chuck inadvertently. Nothing that has been recommended to me stops the motor. When I get things operating again possibly I will post something up here.

RyanH122, 2022-01-17:
We've got about 15" on the ground but it's about done I think. I'll be off to clean that after lunch, LOL!

I ordered a Launch board myself direct from TI and got shipping confirmation it's coming the 24th from Singapore. 

And I really don't know what the problem is... The new encoder would not work on the original pins on the board so that leads me to think it's the TI board. But I'm too nervous to try the original encoder on the second set of working pins on the board to verify in case the encoder is what damaged something. 

Honestly if you have easy access to the TI board and a computer I would be inclined to trying reflashing to board in case the software got corrupted. I kind of wish I would have started there just to see

Chucksbp, 2022-01-17:
I feel the same way, If it is the board, what damaged it. Handling? just a bad board? or the encoder? When I actually have a spare board in my hand I may do some testing. I wish I had the proper equipment and knowledge to test the encoder. 
When I checked TI didn't show any boards at all. My first one i believe came from Singapore via. DHL. it came in about 8 days including holidays.  DHL is very quick.

Chucksbp, 2022-01-17:
Here is a link to some information and possibly a good test for the actual encoder.
https://handtoolsforfun.com/how-to-test-an-encoder-with-a-multimeter/

RyanH122, 2022-01-17:
Thanks for the link. I think I'm going to at least do the multimeter test posted there as that doesn't seem to hard. I don't think I'm going to do any other experiments now as it seems to working good again. Best not to tempt fate, LOL.

Chucksbp, 2022-01-18:
Ryan, Today after snow duty I decided to test my encoder the way the instructions showed. I made an adapter with a 5 volt power supply to isolate it from the board.  To me only one phase was behaving in what i felt was the proper manner. That got me all exited!! so I opened the 300p/r encoder put some terminals on it and hooked it up, Nothing!!!! I expected it to be off because of the P/R count but no it did nothing. So I then removed the TI Launch board brought it to the house changed the pulses to 1,200 and the input to B pugged it all back in and its working. I then took a minute or two and installed the encoder on the lathe and all is working as it should. I checked the RPMs with my Tach the same one as James used. and my RPMs are right on. So I'm stumped as you were, I can say the number one input is bad, and I feel the encoder is bad. I'm defiantly not reconnecting the encoder to be sure!! well any way ill have spare parts. Ill see if the lower P/R is an issue, but if its reading the proper RPMs i cant see how it will effect anything???

RyanH122, 2022-01-19:
Glad to hear you got it working!  Sounds good to me. I'm no expert by any means but I'd think 1,200 p/r would be very sufficient for anything we're doing as a hobbiest. Maybe not on a CNC machine but with a manual more than sufficient. That's still a pulse every 1/3 of a degree right?

If you increased the microsteps of the motor would that offset any loss in revolution resolution??  I don't know how that works either. 

I was planning on testing my "bad" encoder last night but got tied up with other stuff so I'm going to try and do it tonight and I'll update what I find.

RyanH122, 2022-01-20:
Just for grins I did try and test the old encoder as described in the link and got... Nadda. A few times the voltage would jump but that's all. I hope we just got unlucky once with bad encoders and we're all set now!  I did some more threading tonight and the new encoder on the second set of pins seems to be working perfect.

Chucksbp, 2022-01-20:
I agree I am gun shy now!!! although Clough42 hasn't mentioned any issues with his!! he isn't secretive about anything, I'm sure he would bring it up. I may check a new spare i will have in a few days and see what i get from it. I bought a spare 5 volt power supply to have. The one that would not work defiantly only had one phase that worked as expected according to the test procedure. I haven't returned it yet Ill give it one more test on the bench before I do.

Chucksbp, 2022-01-20:
I am also going to soldier up a spare interface board when my hands are steady just in case i Have a second failure. If i have to change the TI launch board i am changing everything electronic at one shot, just to eliminate any chance of failure. The motor is isolated so I really don't think that can be an issue.

Chucksbp, 2022-01-21:
I received my new Launch board and just programmed it removed the tape from the switches, set them and removed the jumpers. I set it up to the standard encoder count and the number 1 set of input pins. I also have a new Dupont connector and strain relief  on the new spare encoder.  I did order a spare Interface board from James. I figured i would do all this while the procedure is fresh in my mind. If the board fails Ill be ready. 
  Also this time I did everything with proper ant static protocol!!! Just in case my Electric personality caused the first issue!!!. 

    I want to thank everyone for their help and input on the problem, and especially Ryan for taking time to start this discussion. with out it things would have been much more difficult. 
    Also a shout out to James for taking time to answer my many questions. I hope he monitors this at times! 
     I also was able to get my  emergency  carriage stop to operate using the Steps- and a normally closed switch.
[image]
[image]

RyanH122, 2022-02-01:
I hope everything is still running good!  I don't want to jinx us, LOL. Mine has been as good as new. I'm glad we were both able to get sorted out. I got a spare ti board myself and I should get an encoder. I think I'll do that on my next Amazon order 😁

Chucksbp, 2022-02-01:
Good to hear everything is fine. My ELS is also operating just as it should. I check it every day even if I don't use it LOL! I'm still a little gun shy I guess. If i knew what caused it for sure id feel better. At this point I have a complete operating second electronic system with exception of the stepper motor. I did a second Clough42 Board to go on the new TI launch board just in case i did have a failure. I was not happy with My first soldering attempt on the board. The second came out very nice. I really think the first encoder was the culprit. Thanks for the checking in post, wish you the best, Chuck

Chucksbp, 2022-07-21:
Just thought Id add an update, its been 6 months and the ELS is working as it should. Its a real treat to use! It makes my 80 year old lathe young again! now I need something for myself!
  I still wonder what caused the failure, its a chicken or the egg question I guess!

RyanH122, 2022-07-21:
I'm not saying a thing but mine has also been happily humming right along ever since. Like you I still have no solid idea what happened but my money is still on the encoder going bad and taking the one set of connections on the ti board with it but with everything working so well I'm still not willing to swap connections and check, LOL.

gusosborne, 2022-09-30:
Just went to use mine for the first time in about a month yesterday and I seem to have the dreaded dead port 1 as well. I already had a new spare encoder on hand, no dice. Will be re-programming this weekend and I’ll report back. 
Would be nice to know what’s causing this- or at least a good way to protect them. Perhaps some type of opto-isolator module..?

Chucksbp, 2022-09-30:
Sorry your having problems. let us know what happens. What brand encoder did you use?

RyanH122, 2022-09-30:
Sorry to hear you're having this issue. Please keep us updated. I'm also interested in what encoder you have been using. I (knock on wood) haven't had any more issues since but still have no clue exactly what happened. I'd definitely be interested in what if anything you find out and if you come up with a plan to prevent it from happening in the future.

gusosborne, 2022-09-30:
Thanks guys- I was using the same “omron”, originally linked in the wiki. My spare / backup is this: https://a.co/d/1p6Eouu

Not sure if I should trust either encoder now. Is that the consensus, that the encoder could somehow damage the port? I guess I can’t quite wrap my head around it. 

A weird thing about my situation is the stepper moves slightly when the encoder moves - just little clicks and barely noticeable- same with either encoder. Some minor signal is making it through to the stepper. The RPM readout never registers anything though. 

I also ordered a new boost board, just in case. Im considering going for a higher quality encoder or building some type of bearing support as mentioned above. My old encoder still looks and feels brand new. 

Would something like this be useful for protection at all? I can’t really see how it would be, other than ESD protection:
https://a.co/d/634F0AV

Turner75, 2022-09-30:
> Thank you- I went with your suggested higher resolution encoder. I also
> ordered 2 spare TI boards.
>
> I’m pretty sure the product page has changed for that encoder link I sent,
> because the spare I have definitely worked during my initial setup and
> testing.
>
> Thanks again for all the help!
> Im going to attempt to reprogram the board now.

Having built and operated the ELS on my Grizzles for around nine months
now, and had very little trouble after some initial problems with the
interface board, I offer the following.
   It seems that a lot of the problems stem initially  from the interface
board, as was my early problems, dry solder joints and a diode that was
inserted wrong way round. Encoder problems fall into two categories;
Wrong specifications, voltage, interface configuration and differences in
signal phasing and incorrect mounting.
        The second category is quality of the encoder. Some of the less
expensive offerings on the Web are not upto the requirements of a control
system that can run at upto 2,000+rpm. The bearings and case rigidity are
not sufficient to give lasting performance. Any bearing wobble or shaft
flexing will give intermittent counts and eventually total failure as the
clearance between the optics and the encoder disc is very tight, resulting
in the photo print on the disc getting wiped off.
My Swiss Baumer 1,000 count encoder was pulled from a previous installation
after about five years continuous operation and is still running fine even
though it is driven at twice the spindle rotational speed.

clough42, 2022-09-30:
It looks like the link on the GitHub page still works, though there are a
lot of different, very similar encoders available.

The one I use and recommend is the Omron E6B2-CWZ6C.  The "6C" in the part
number indicates that it's NPN Open-Collector output, which is what you
need.  I recommend the 1024 P/R resolution when running 1:1 with the
spindle.  More resolution than this is unnecessary, and less than this can
lead to rough running of the leadscrew motor--especially for large thread
pitches when using gear or belt reduction between the motor and the
leadscrew.

James

[4 more replies]