clough42/electronic-leadscrew #198: Rapid feed

KennethLuscombe, 2022-01-22:
I have purchased everything for the lead screw Stepper motor project and about to put it together.
Question
Can I get the stepper motor to do a rapid feed by pushing a button with the current setup.
Can I connect a PWM generator with a button for rapid into the PUL + PUL - in parallel to the TI board connection.

JonWoellhaf, 2022-01-22:
Sorry, I believe the answer to both your questions is no. Rapid is certainly something that could be easily added.Sent via the Samsung Galaxy S20 FE 5G, an AT&T 5G smartphone

pmelevende, 2022-01-23:
That would be a handy feature!

PaulCElliott, 2022-01-24:
You could do it using three switches. 
1)A 2P2T to route the PUL and DIR to the motor.
2) a "monetary close" to connect the pulse generator (555 maybe?) 
3) n/o to select direction via a couple of resistors. 
Its not very elegant and rather spoils James's clean design. Obviously it couldn't be used when threading as the processor would lose sync. You would have to set the generator to a convenient speed or add a pot. 
I tried it with a pulse generator on the bench and it seemed to work. The ELS just picked up when switched back in; but clearly with a different sync between leadscrew and spindle.
Be nicer in software though.

KennethLuscombe, 2022-01-24:
Thank you Paul

Hi James
Be nicer in software though.

closed 2022-01-26 by KennethLuscombe