clough42/electronic-leadscrew #201: Electronic leadscrew metric

chrisjan85, 2022-02-06:
Good day James. 

Please are you able to assist me.

I am struggling to get my Leadscrew sorted out. With the coding of the correct hmm setting and microsteps

I have the following components:
Your booster pack
Texas Instruments Microcontroller Board LAUNCHXL-F280049C
Omron E6B2-CWZ6C 1024 CPR
Nema23 3Nm iHSS60 closed loop Stepper Servo motor 36VDC with factory fitted driver ( attached Image and microsteps chart for dip switches)
My Leadscrew Major Diameter 20mm

I currently have the Leadscrew Hmm at 400
Encoder at 1024 CPR
Stepper resolution at 1600.
Stepper microsteps at 1

If I set the vernier caliper to 1in/25.4mm it counts 13 tpi on the Leadscrew threads.
 
Should I rather have it set on the code at 13 tpi or must it be set in metric?
Secondly: If it must be set on metric what feed/thread rate will give me 1:1 cause I don't get it to run 1:1.
Thirtly: Am I using to much microsteps should I rather use 800 microsteps? 

Please if it is possible to advise and help it would be much appreciated.

Thanks
Chris Oberholzer 
South Africa 
[image]
[image]

Chucksbp, 2022-02-07:
First of all the encoder count should be 4096. its 4 time the pulse per rotation. What is your stepper drive motor ratio you have it set up for a 1 to 1  ratio??? with 1 micro step. Also I'm not familiar with the metric lead screw set up but why do you mention diameter??? You say you have it set at 400, that would indicate a 4mm lead screw pitch. Using your thread pitch gauge measure your actual lead screw thread pitch.  you are only concerned with the pitch.  the stepper resolution is set to what ever you set the motor dip switches to. . as long as its the same I don't think it matters. The stepper micro steps is set to the drive ratio of the motor to the lead screw. if you are direct its 1 if you 2 to 1 you use 2 3 to 1 you use 3 and so on. 
As far as to check and get a 1 to 1 ratio of the spindle to the lead screw you have to set the ELS to thread and the pitch is the same as your lead screw pitch If everything is set up correct it would then run 1 to 1 I hope this helps.

chrisjan85, 2022-02-07:
Good day Chuckspb

Thanks for your response. 

I made a mistake on the encoder count, for some reason the 1024P/R printed on the encoder got stuck in my head. I will change that on the Controller. 

All my drive ratio is 1:1 from vfd to spindle, from spindle to encoder, from stepper to leadscrew. So then the stepper dip switches so be then on default all 4 switches set to on?

I feel actually like a donkey for some reason i tried figuring out the Leadscrew metric pitch but never thought of using the pitch gauge. Thanks for dropping the penny 🙈🤣. 

And another opinion request? 
What should be the best to drive the vfd to the spindle timing belt or v belt. I am still in progress with upgrade so currently it is looking rough, but see attached picture of current setup. 
[image]

Thanks for your assistance I believe i should be sorted for now.

Regards 
Chris

Chucksbp, 2022-02-07:
First off you have to use toothed timing belts for everything. It has to hold exact position with no slip. The drive should be the larger HTD 3m by 10mm drive belt and pulleys. 
  I'm confused you say you have all 4 switches set to on. That shows to be some sort of default setting. You then state you set the steeper resolution in your configuration H file to 1600, according to your motor chart that would be ON OFF ON ON that also will throw your motor speed off.  
  it also looks like your encoder belt may be loose. even the cog pulleys will jump a tooth if not tight enough. also MIC your 1 to 1 pulley diameters. I found that the Chinese pulley diameters are not all the same I had problems at first my encoder was gaining position at first. it was slow but in 5 minutes of run time it would gain 
. I don't know how much this small amount would affect a thread but i was able to correct it by filing a few throw off the larger gear. I would think a toothed gear would stay in time but it was not. 
    When I did my programing I tried to keep as many standard settings that showed on the H file because at that point i didn't understand as much. but after having a failure of the encoder input on the launch board shortly after it was working, with the help of others and re watching the videos a bunch of times, I'm starting to get a better understanding.   
   It defiantly can be quite overwhelming at times.   I will say once its set up properly its great.