clough42/electronic-leadscrew #203: ELS, Control Panel how to change start setting,  starting from metric and not Imperial

andrea271068, 2022-02-17:
Good day, James.

Please are you able to assist me.

In the control panel, when I switch on, I would like to start from metric and not Imperial, but because I don't have experience and competence in programming I'm scared to corrupt the code. 

Surfing on, I found that maybe this option is placed in: userinterface.cpp

There is this string: this->metric = false; //start out with imperial

I don't know how to change this input, can you help me, please

jimhinze, 2022-02-17:
I haven't compiled the code yet, but changing that to this->metric = true;  will default it to metric from the start. It's not set to an initial value in the header, so this is the first place the value is initialized. I'd also change the comment to //start out with metric  so you know what you did.

It won't corrupt the code, it's a one line change.. so if I'm wrong, you can just change back and re-compile, or wait till someone corrects me :)

andrea271068, 2022-02-17:
this in line 79? I should type: this->metric = true; start out with metric  
instead of:  this->metric = false; start out with imperial

jimhinze, 2022-02-17:
yes, based on what I see and my understanding with of the code. Bear in mind, I've not compiled it yet (I'm one of the ones waiting on TI boards to be available). So, forgive me if I'm wrong.. someone will correct me :)

andrea271068, 2022-02-17:
Thanks, let’s try

Andrea Pisani
Overseas markets export manager
Mob: +39 335 70 98 102 - Ph: +39 0375 783687
Mail: ***@***.******@***.***>

Composad srl – Italy

[http://www.grupposaviola.com/intranet/loghi/composad-logo.jpg]

COMPOSAD s.r.l.

Sede Legale: Viale Lombardia, 29 - 46109 Viadana (MN) - Italy

ph: +39 0375 783611 | fax: +39 0375 7836695

Sede Operativa: Via Lombardi 1 - Z.I. Gerbolina - 46019 Viadana (MN) - Italy

C.F - P.IVA/VAT code e R.I (IT)02317920201

www.composad.com<whttp://www.composad.com>

[http://www.grupposaviola.com/intranet/loghi/piede.jpg]

Il presente messaggio di posta elettronica e i relativi allegati sono strettamente confidenziali, soggetti alle leggi vigenti, pertanto se ricevi questo messaggio di posta elettronica per errore, per favore contattaci: e' assolutamente vietato usare, copiare, o divulgare il messaggio e ogni informazione in esso contenuta, tranne che al destinatario originario. Le comunicazioni Internet non sono sicure e non ci riteniamo responsabili per eventuali abusi di terze parti, per alterazioni nel corso della trasmissione, né per danneggiamenti causati da virus o altre anomalie a noi non imputabili.
----
This message and its attachments are addressed solely to the persons mentioned above and may contain confidential information. If you have received this message by mistake, be informed that any use/copying/forwarding of any part of its content is completely prohibited. Please return it immediately to the sender and delete the message. Internet communication is not secure and therefore we are not responsible for any misuse of it by third parties; Alterations during the transmission or damages caused by viruses or other anomalies are not attributable to us"

'

jimhinze, 2022-02-21:
Andrea,

did it work?

> Thanks, let’s try Andrea Pisani Overseas markets export manager Mob: +39 335 70 98 102 - Ph: +39 0375 783687 Mail: ***@***.******@***.***> Composad srl – Italy

andrea271068, 2022-02-21:
Yes, many thanks, was very easy 🙂✋