clough42/electronic-leadscrew #204: Encoder connections

KennethLuscombe, 2022-02-20:
Hi.
The encoder i have has only 4 wires and a sheild  - part number (B38S-06G5-26C600B / 180744)
red
black
green
white
sheild
it looks like i need to have 5 wires - will this encoder work on the TI board

I was thinking this:
red pos
black common
green A
white B
it looks like there is one wire missing?

Please advise as to the actual connections to the TI board

KennethLuscombe, 2022-02-20:
EQEP1I maybe missing - is it needed?

Turner75, 2022-02-21:
HI Kenneth
I built the ELS, all up and running now, great! I used a Baumer encoder,
500 ppr, one of two I had to hand. I found that just the A & B quadrature
outputs worked fine, the complements -A, -B , not required,  nor is the
single pulse per rev o/p the fiducial marker pulse. My encoder is coupled,
1:2 gearing to spindle giving 1000ppr.
BTW, seems like quite a few swarf makers have had trouble with encoders
going bad. I suspect it could very well be in some cases too much side
loading on the encoder shaft. They are not built for excessive side loads
so the bearings or housing fail and the spinning photo patterned disc
generating the pulses catches the opto assembly and scrapes off the pattern.
Just my two penn'oth
Cheers Graham

KennethLuscombe, 2022-02-21:
Thanks for the reply.
The only wires coming out of the encoder are Red Black White AND Green.

So
Does this sound correct

Red Pos
Black Neg
Yellow A
White B
Nothing connected to Z or I

The encoder is a 600ppr

so
Setup of encoder in the software 2400 ?

Thanks
Ken

Turner75, 2022-02-21:
HI Ken
I don't know whether the wire colour coding is standard across all rotary
encoders, unlikely. If you can't  find the connection data for this
encoder, you could try testing it out if you have some knowledge of
electronics and test equipment such as a DVM, an oscilloscope and an
adjustable bench power supply. Having said that, testing it out with just a
cheap DVM with no power on the encoder can tell you quite a lot.
    DVM (Digital Volt Meter).
 NOTE* All the following tests are done with NO power applied to the
encoder. Outputs from encoders are generally open collector, so, checking
with a meter on ohms range should give high or open circuit readings.if
your DVM has a check diode setting, then you should see a reading with the
meter leads connected one way round and open circuit in the other
direction. In contrast, the power supply inputs will usually give some
value of resistance in ohms in both directions and will vary depending upon
which way round the meter test leads are.
Hope this helps you out.
Cheers Graham

KennethLuscombe, 2022-02-23:
The encoder is working with the 4 wires connected. Thank you

closed 2022-02-23 by KennethLuscombe