clough42/electronic-leadscrew #209: Lead screw settings

KennethLuscombe, 2022-02-25:
Hi. I have tried many settings and cant seem to get it right.

I am trying to get the lead screw to turn one revolution when the chuck one revoluton. I set the dislpay to mm and the thread to 1.5

the lead screw is 1.5mm
the encoder is a 600 
The setper motor is a nema 17  1.8 with encoder has a pully half the size on the steper side than one the lead screw side.

Please spell the settings out. I have tried many settings and its close but no right.

The settings i am using are :
#define LEADSCREW_HMM 150
#define STEPPER_MICROSTEPS 28 ( I have tried from 2 - 28 and 26 - 28 is close)
#define STEPPER_RESOLUTION 200
#define ENCODER_RESOLUTION 2400
I have changed nothing else
also how do i make it start up in mm not inch

Please help

KennethLuscombe, 2022-02-25:
using software version 14

KennethLuscombe, 2022-02-25:
As per the above I have just tried

stepper resolution 1000
stepper micro step 2

I then get 3 revolutions of chuck to one on lead screw.

KennethLuscombe, 2022-02-25:
if I change the encoder to #define ENCODER_RESOLUTION 600 all is well except the RPM is on display only show 1/4 the actual RPM

Chucksbp, 2022-02-25:
I would say the lead screw is correct at 150hmm, the encoder resolution looks correct at 2,400 assuming your encoder is operating at a 1 to 1 ratio. What is your stepper motor switches set to. Its best to set it a 1,000 and then your resolution would be set to 1,000 and the micro steps should also be set a 2 for a 2 to 1 ratio. 
One thing I found out during my installation is all pulleys are not the same size. I had two 60 tooth pulleys of different brands driving my encoder one was a bit larger diameter and although they were both 60 tooth they would get out of sink very quickly. Even the same brand cheap Chinese ones will be off from one batch to another. I finally found two very close and took .003 off the diameter to make a matched set and i haven't had an issue.  Put a mark on both the encoder and drive pulley and run it for a few minutes and see if the marks are still in sync. Also belt tension seems to play a roll on the encoder

KennethLuscombe, 2022-02-25:
I feel silly . Thank you so much. I missed the fact that i could set things up on the stepper driver. I had it set to 8000 so 1000 sorted it out.  It is working so perfectly now. Thank you so much.

and thank you Clough for building such a nice product.

closed 2022-02-25 by KennethLuscombe