clough42/electronic-leadscrew #210: Spiral effect on work surface

andrea271068, 2022-03-09:
Ho, please somebody knows why there are spiral effect of the worked surface? 
Everything look work properly, even the threading are precise, but I have this horrible not even surface on the pieces, spirals streak. May is something related to the driver step configuration?  

Please some body can explain and maybe drive mw to the solution, thanks

dsgroninger, 2022-03-10:
Andrea,

I don't think that there's anything particularly wrong here, just not the correct combination of speeds, feeds and cutter for that particular steel.  That carbide cutter appears to come to a sharp point, where a little radius might make for a smoother surface finish.  You may find and an old fashioned cutter ground from high speed steel with a tiny radius at the cutting point might do a nicer job.  If you don't mind turning some of that material into wasted chips, I'd just try some different speeds, feed rates and cut depths to see what works best.  There's also consulting the data sheets for the cutter you're using and see what they recommend for speeds and feeds.  I don't have the experience to tell you a good stating point, but I've been surprised at the variations in surface finish that I've gotten from 1018 cold rolled steel bars just from different depths of cuts and speeds.   Machinery's Handbook will get you in the neighborhood, and just experiment a bit from there.  Keep in mind too that some steel is just crap.  I never get a mirror finish on the 1018 from my local sources.  I hear that 12L14 steel is the hot ticket for getting really nice results on hobby sized machines, but I haven't tried it yet myself.

Good luck, and have fun making chips!

Dan

andrea271068, 2022-03-10:
Many thanks, ai will try different fead and speed, let's see, I had the impression too that maybe is the material too, thanks I will let you know.

MorryStu, 2022-10-28:
Yes, speed and feed are your friend, along with the cutter. I use tool steel cutters and grind by hand. A round nose cutter makes a really nice finish

The steel also helps...cold rolled steel is "gummy" and sometimes will just not play right

Thats why emery was invented