clough42/electronic-leadscrew #212: ELS working - but no reverse

IamTomzo, 2022-04-02:
Greetings,

I have my ELS working but for some reason it won't go into reverse.   I thought it might be a bad connection so I replaced the round soldered connectors that Clough42 used with panel mount screw terminal connectors - those little round connectors are finicky.

Anyway, all my connections are now good and the system works, but still no reverse.   It had worked in the past, but at some point stopped.   I am using the iSV57T-180 integrated servo and can reverse it by flipping the dip switch on the end, but I figure that is not the type of switch that should be flipped a lot.

I was wondering if anyone else had experienced this or had any advice on where to look for the problem.

Thanks

Tom

IamTomzo, 2022-04-03:
I should add that the LED on the display changes from forward to reverse when the button is pushed.  I am wondering if this function is handled by the display board, the Clough42 board, or the TI board.

dsgroninger, 2022-04-03:
Tomzo,

If the reverse button toggles the LED, that's an indication that the button on the display board works, the signal has been sent to the TI board and the TI board is responding to the switch closure and trying to perform the reverse function.  The display board connects to the TI board with a serial bus, with all the switch actions and display actions communicating over that single bus.  So I'd say that all looks good.

Now, I'd look at the connections between the TI board through James' buffer board to your stepper motor driver.  There are three signals there - step pulse, enable and direction.  If your setup turns the stepper in reponse to the spindle turning, then the enable and step pulse must be working.  Check the reverse signal - you should be able to check it with a meter if you don't have an oscilloscope.  It should  be logic high in with the system in one direction, push the reverse button and it will be logic low.  Low would be ground, high either +5 or +3.3 volts, I forget which on this system.  If that signal doesn't change state when toggling between forward and reverse, it could be a fault on buffer board from James, or possibly back to a signal on the TI board.

If the signal changes state but the stepper still doesn't reverse, then there is a fault in the stepper controller, or possibly there's a setting on the stepper controller that is set incorrectly.  Check the instructions for the stepper controller and see if there is anything disable reversing and see how that's set.

Good luck debugging,

Dan

IamTomzo, 2022-04-04:
Thanks Dan - that sort of bread crumb is exactly what I needed!

IamTomzo, 2022-04-04:
As usual the problem is me.   When I reworked my wiring I inadvertently landed the DIR- wire in the wrong spot - duh.   Sometimes I wish I could just go back in time and slap myself.....

closed 2022-04-04 by IamTomzo

JonWoellhaf, 2022-04-04:
Ha, ha! Join the club. Glad you worked it out.

From: IamTomzo 
Sent: April 3, 2022 20:00
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] ELS working - but no reverse (Issue #212)

As usual the problem is me. When I reworked my wiring I inadvertently landed the DIR- wire in the wrong spot - duh. Sometimes I wish I could just go back in time and slap myself.....

dsgroninger, 2022-04-04:
> As usual the problem is me. When I reworked my wiring I inadvertently landed the DIR- wire in the wrong spot - duh. Sometimes I wish I could just go back in time and slap myself.....

Good deal, glad you found it!