clough42/electronic-leadscrew #213: How to change Feed direction, Threading is OK.

AZ-Ron, 2022-04-03:
Hi All... Quick background, I started the ELS project along with James Youtube tutorials. Got to the point of bench testing, 
then uprooted myself from AZ and moved to Idaho. Finally have my new shop up and running and got back to the ELS. Got it installed on my Logan 6560H (Vintage 1965, 14x30). Not considering the size difference with what James has, I purchased the Closed loop stepper and driver that he used initially. Once I dived back into the project, I knew that that setup was inadequate for my lathe, but hooked it up and just did some movement test. Everything worked great. Ordered a new stepper/driver from StepperOnline, and have it all set up and working perfectly, with one minor glitch.
The Logan uses a lead screw of 8TPI, that has a longitudinal groove running the length of it. The groove is keyed to a worm gear in the apron which provides power feed to the carriage and the cross slide. Took a bit, but I was able to get the different ratios figured out and the Threading and Feed modes now provide the correct amount of travel per rev. The issue, granted it's a minor one, is that when the control panel is in Forward, the threading runs in forward, but switching to Feed, the carriage moves the opposite direction.
I know  how to change the direction in Config.H but that seems to change the direction for both Threading and Feeding. Is there a way to just flip the direction in Feed? I've searched High and Low and have found answers to every one of my questions but this one.
As a side note, the Feed ratio is pretty low on the Logan, requiring a ratio of 22.74 to obtain .005/rev at default. This is with a 3:1 ratio between the stepper and lead screw. I have a 2:1 set of pulleys on the way. From my reading, 22.74 will tax the stepper at higher feed rates and spindle RPMs, but I think the 2:1 setup with keep everything happy and in the Pink. 

Thanks all... This has been an awesome project and I'm really happy with the results. James, if you happen to read this, I can't thank you enough for your videos... ELS and the others... really enjoy them. Very well presented and I always learn something!

Cheers!

Ron

Chucksbp, 2022-04-30:
Why not just use the reversing button on the Clough42 display interface. You can change direction any time the lathe is stopped.

1jdupton, 2023-02-07:
> Why not just use the reversing button on the Clough42 display interface. You can change direction any time the lathe is stopped.

Mine is running one direction no matter which way the encoder turns and forward reverse button does nothing. I can't think of the answer but it was switching with the spindle direction last night.

1jdupton, 2023-02-07:
I found my problem being a loose wire on the driver connector where you jump three grounds.