clough42/electronic-leadscrew #214: Will Enable Work With a DMM DYN2 Servo Controller

frostheave1960, 2022-05-01:
Hi All,

As shown below, Enable (Pin 11) is Positive and has a Negative Common (Pin 7).
The diagram for the ELS PCB shows a Positive Common.
Will it be possible to use Enable with this drive?

Thanks for your help!

 
[image]

pmelevende, 2022-05-04:
HI Frostheaven,

Yes - Check 'Direction and Enable' on this page https://github.com/clough42/electronic-leadscrew/wiki/Configuration-Settings

Thanks
Pauly

>

frostheave1960, 2022-05-04:
Very good.  Thanks Pauly.

And just to be sure, I won't be doing any damage to the hardware by physically reversing the polarity to the Enable lines,
Correct?

Thanks again
Bob

pmelevende, 2022-05-07:
HI again Bob

No you will be fine. Even if you mess up somehow, I'd assume that 270 ohm resistor on the enable line is there to protect you :-). Having said that , if you are worried, check the polarity before you connect it!

Thanks 
Pauly

>