clough42/electronic-leadscrew #215: Thread and feed rates are to fine

dbike199, 2022-05-16:
I have been doing the G0602 VFD and ELS conversions by James videos. All has worked well. I finally received TI LaunchXL-F280049C last week and programed the board. The only issue I have is the threading and feed rates are way to fine. The best I can tell they are 3 times to fine (clue). I played with the setting in Code Composer 9.1.0 but I can't seem to hit the correct combination Stepper Microsteps and Stepper Resolution for my setup. I'm pretty sure that is where my issue is. 

I ordered this stepper and driver setup from Automation Technology https://www.automationtechnologiesinc.com/products-page/hybrid-servo-system/hybrid-servo-drive-kl-5080h. But received this Driver https://mecheltron.com/en/product/es-d508. All the manuals and specs are on this link. The kit came with a "NEMA23 KL-23-2N-1000Hybrid Servo Motor with one Encoder 1000 line".

I believe the issue is with the 3:1 ratio of the Hybrid Servo and Lead screw setup

A few facts on my setup:
G0602
3:1 ration from Servo to Lead screw via pullies (24t and 72t Pullies)
G0602 is set with a 1:1 ration on transmission (1 and A)
Omron E6B2-CWZ6C Encoder. I believe it is programed correctly as my spindle RPM is very accurate compared to the display when checked with a tachometer.
All the buttons and display work as they should. 

I tried the Trouble shooting recommendations per James Github page but have not solved the issue yet.

As stated above, the issue seems to be with the correct numbers for the Stepper Microsteps and Stepper Resolution for my servo and driver setup.
[image]

Any suggestions would be great welcomed.

Thanks,

Mark

Chucksbp, 2022-05-16:
If you are running a 3 to 1 ratio at your servo motor, your stepper micro steps should be set at 3  the stepper resolution should be set to what ever you have set your dip switches to on your controller. 
I have my stepper resolution set to 1,000 on both the controller and in the software and also have a 3 to 1 ratio and have 3 in the micro steps and all works well.

dbike199, 2022-05-16:
The ED-D508 only has 2 dip switches, and they are for direction and a self test.
[image]

It states the Default Resolution is 4000.
If I set Micros steps as 3 and Resolution as 4000 I get several errors with the 4000 on the build. I tried flashing and received this message
[image]

Chucksbp, 2022-05-16:
Im not an expert but from my understanding the end result of speed is resolution times micro steps. I don't know what the limit is in the software. But your motor is set up for 4,ooo steps per revolution you need 12,000 steps to make your lead screw turn at the correct speed, so you need to try some combination the software might except. being that it will accept 8 micro steps try a resolution of 1,500 that would get you the 12,000 value. 
the only other thing you can do the drive states the micro steps can be refigured by software, you would have to contact the driver company for the software and cord to achieve that. 
the other would be if your motor has enough power change to 1 to 1 ration.

Scubamachi, 2022-05-16:
I had a similar issue with that stepper motor, as it is only 1.2 degrees per step, and not 1.8 steps like James used. I wound up getting the right motor after wasting alot of time, playing with ratios.

Scubamachi, 2022-05-16:
Check out my issues in Github (69 & 70) as it might help, thanks.

dbike199, 2022-05-16:
Chuckspd might not be an expert, but in my eyes, he is pretty darn close to an expert. His recommendation of 8 micro and resolution of 1500 was on the money. Here is a picture of a 20TPI thread I did really fast to see if it was correct.

[image]

I will play with it a bit more to verify accuracy.

Thanks for the help Chuckspd and Scubamachi!

The G0602 is a joy to work with now with the VFD and ELO. Thank You James!

Chucksbp, 2022-05-16:
Your welcome!! i'm glad that combination worked for you! I had no idea that the ELS soft ware was limited to 2,000. But the work around did the trick. The ELS is a great addition to any lathe with out quick change gear box. I love mine on my Atlas 618
[image]

electroAlain, 2023-08-29:
Hi my problem is similar. I need to cut threads of multi entry 1 TPI.
But even at 8 TPI The TI control trips when my NEMA 34 12N 100 step motor reaches 40 rpm
I looked all over the program and I cant figure it out

Chucksbp, 2023-08-29:
At 1TPI that motor would have to fly the lead screw would be spinning much faster than the spindle. I'm not sure that is possible even if you ran a 1 to 1 ratio on your stepper motor. I know mine screams at 8TPI that's the fastest I have cut at.