clough42/electronic-leadscrew #216: Rotary encoder voltage issue

blackfister, 2022-05-26:
Hi, I have purchased de omron E6B2 rotary encoder.
Sadly I got the CWZ5B, which runs on 12 /24 dc, not the 5 or 5 /12.
How could I bypass this issue? If I use an external power supply 12vlts, how should I connect to j12 port?
Thanking you in advance
Martin

Chucksbp, 2022-05-26:
I made the same mistake. I contacted the ebay seller and they gratusoly exchanged the encoder for the correct one. Being that you are picking up the voltage from the launch board id be very cautious making any changes. that board seems to be a bit delicate in that area. I had a failure in the number 1 circuit of the encoder with the proper encoder. I don't know if the encoder failed or the board failed. but i replaced the encoder and reprogramed the board to use the second encoder input and its been fine.

blackfister, 2022-05-26:
Thanks for responding, yes I see that I might have to change to proper voltage encoder

closed 2022-05-28 by blackfister