clough42/electronic-leadscrew #220: Display changing... Back and forth

blackfister, 2022-06-02:
Hi, all I managed to get display and tachometer, but I get "clough42" and "els 1,4" appearing and disappearing, any hint as what to check for?
[image]
[image]
[image]

Chucksbp, 2022-06-02:
The only thing that i could think of is that some where you are intermittently loosing power do the boards. That is only displayed at start up. Why do you have the Jumpers right above the mini USB still plugged in on the main board? you are supposed to remove those so that you don't get power from two sources. What happens if you unplug the mini USB from the board. 
If that doesn't change anything I would look closely at all your soldier joints related To both the 3 volt and 5  volt power on the Clough 42 interface board. https://raw.githubusercontent.com/clough42/electronic-leadscrew/master/docs/images/Continuity/ELSBoost-Power-Continuity.jpg

Chucksbp, 2022-06-03:
That's good the wire was blocking the complete jumper in your pictures, so i thought I's mention it. God luck.