clough42/electronic-leadscrew #223: Halt on "WAIT_FOR_SERIAL;"

meckarmulle, 2022-06-15:
Hello
Just start first test on my ELS project, Thanks James, rely fun to play whit this build.
I run to a problem that I guess is because of the of the encoder but I'm not shure.

I start the progran in debug mode and it seems to stop at “WAIT_FOR_SERIAL; “ Line 71 in CPIBus.cpp

Information on display is freezing,  

Anyone that seen this before and knows what the problem is
[image]
?

Chucksbp, 2022-06-21:
I'm not absolutely sure, but I would be leaning to a bad display board or possibly a poor connection at one of your crimps or soldier joints. I would first download and print the Trouble shooting guides and closely ohm out the connections with the wiring leads in place. Then I would try a second or third display. The display boards have a high failure rate out of the box. I had one that did a similar display as I remember. 
Even with no active encoder the display after being programed should start with the Clough42 display then software version then should have 0 RPM and .005 for feed rate. the top lights lit should be power, FWD, Feed, and inch. you should then be able to toggle the feed rate up and down.

meckarmulle, 2022-06-21:
Good to know that it will start whitout encoder, then almost only thing left is the display board, as you say. I have ordered severel new display board

Thanks for answer

Chucksbp, 2022-06-21:
I thought i had damaged the board changing to a right angle plug but that wasn't the case, I believe it was just a bad board. I did find that you can car5efully bend the pins to a 90 and install the plug on the end with a bit of hot glue to lock it in place. it will just barely fit in the box.  
  this is what the display should look like with no encoder input. 
[image]

meckarmulle, 2022-06-30:
Hi, I have now received 4 new display boards but I got the same result on every one of them. I opened my spare TI board and tested that one, same result. 
I watch the part 10 youtube video very carefully and it don't start with the text “Clough42”, it start with the text on my picture.
Another different from video is that my TI bord start with light on LED1, LED3 and LED5 before I load cod. 
After I load cod I have light on LED1, LED2 and LED3
After I have started the cod LED5 start flashing. 
Could it be some mismatch between TI board and  Code Composer Studio?
I using  Composer Version: 9.1.0.00010 and TI board part number LAUNCHXL-F280049C 3002056395.

meckarmulle, 2022-06-30:
I just see another strange things, on your picture, I can se that there is i chip on Clough42 board position U2, my board dont have any chip on that position :-)
Of course, that is the reason for my problems. Can anyone tel me the chip label? 
[image]

Chucksbp, 2022-06-30:
Yes You are defiantly missing a component on that board!! I would be contacting Jim for a replacement. 
[image]

JonWoellhaf, 2022-06-30:
According to the BoostXL schematic, U2, the chip in question, is an AT25160B-SSHL-T.

Alas, according to https://www.findchips.com/search/AT25160B-SSHL-T it is apparently in stock nowhere on Earth at present.

I suggest you contact James.

Regards,

Jan (Yan)

From: Chucksbp 
Sent: June 30, 2022 13:27
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] Halt on "WAIT_FOR_SERIAL;" (Issue #223)

This is the best I can do with the chip label, I'm 72 years old and with two sets of glasses and a magnifying glass i still couldn't read all the numbers. this is a picture with a magnifying app, I still would just contact Jim and get a new one. I'm sure he would replace it no charge!

meckarmulle, 2022-06-30:
Thanks, yes I have send him a message true eBay, hopefully that only my board missing a chip, not so good if it is a hole bach

clough42, 2022-06-30:
The chip is indeed a Microchip AT25160B-SSHL-T, but it isn't necessary for
the ELS.  I included it in the design originally, just in case I wanted it
at some point in the future, but never ended up using it.  I've been
including it on the board because it was cheap, but as Jan pointed out,
it's not available right now for any price.

There should have been a card in your package explaining this, but maybe i
missed one.

As usual, I tested all of the boards with the missing chips, and everything
else worked properly.  I also installed one of the new boards on my lathe,
and have been using it, just to be absolutely sure there isn't an issue.
This doesn't totally rule out the possibility of a problem with your boost
board.  ESD damage to one of the transistors could cause issues like this.
I've only ever seen that once, and I detected it while testing.

Is the debugger stopping on the WAIT_FOR_SERIAL line by itself, or is that
just where it always ends up if you stop it manually?  The firmware
transmits a word of data to the display, and that line of code is just
waiting for the interrupt to say that the bits have all been sent.  It
shouldn't require any response from anything externally, but the serial bus
is the slowest thing in the system, so that's where it's spending most of
its time, and that's where it'll usually be if you break in.  If it's
stopping there automatically, that's something I've never seen before.

I would still lean toward an issue in the wiring of the control panel, a
crimp issue, a bad solder joint, a bridged pad on the PCB, or something
like that.  There are detailed continuity checks described in the wiki.

James

meckarmulle, 2022-07-01:
Hi, I found the note abut the chip, when i saw it I remember that I have read it, but under troubleshooting forgot about it, my apologize  for that . 
Back to start again.
Program stops almost immediately whit LED5 fast flashing, no response on buttons.
I have 2 TI boards, same result on both
I have 5 display board, same result on every one of them. 
I have checked every strip on the Clough42 board, no  problems there.
Next step for me is to hock up my oscilloscope and follow the signals.
I watch the part 10 youtube video very carefully and it don't start with the text “Clough42”, it start with the text on my first picture.
Another different from video is that my TI bord start with light on LED1, LED3 and LED5 before I load any cod.
After I load cod I have light on LED1, LED2 and LED3
After I have started the cod LED5 start flashing.
Could it be some mismatch between TI board and Code Composer Studio?
I using Composer Version: 9.1.0.00010 and TI board part number LAUNCHXL-F280049C 3002056395
When I connected TI board for first time it want to upgrade the debugger, if I answer No it cant find the debugger, if I answer yes it upgrade the debugger and then it load software whiteout any errors.
BR
Perry

meckarmulle, 2022-07-03:
Hi
Here is an update of my problem:
I triple checked the Clough42 board and it works perfectly fine. Then I couldn't see any other things to test then start all over again. I deleted Composer and all software, download everything once again and start from beginning. Very, very carefully follow the instruction.  
And now it works.... I dont know what I did wrong before but I am very happy that it works now. 
Well, I haven't start test with encoder and step motor yet, I think I hang on to this feeling some time before I start with that :-)
Thanks for support, maby this can help any other guy that run on same problem as I did.
BR
Perry