clough42/electronic-leadscrew #224: ELS 1.4 loses thread synch when changing direction

gregreid, 2022-07-29:
First of all, the obligatory but very sincere THANK YOU JAMES for all of your hard work on this project and for sharing it with the world.  I've learned a great deal in following your excellent detailed videos and had a lot of fun building mine.

On my Chinese MX-210V mini-lathe, I don't have a dial/index on my lead screw.  So whether cutting metric or imperial threads, I need to leave the leadscrew engaged, back out the cross slide, and reverse direction after each thread-cutting pass.  No big deal once you have the muscle memory.  But...

I've found that it will sometimes lose thread synchronization for the next pass, messing up the thread.  I spent quite a while making sure that everything is "tight" in the lathe and leadscrew drive.  Nothing is slipping.  I spent a while longer ensuring proper shielding and grounding of the encoder and servo motor drive 'coz I'd assumed that the problem was most likely some lost pulses. 

But I've found now that the REAL problem is when switching direction -- and of course I need to do this frequently when cutting threads on my lathe, lacking a thread dial/index.  I found that I can now fairly easily recreate the problem so made and uploaded a video showing my setup and the problem recreation:  https://photos.app.goo.gl/ri2bKXaHttoGKzpX7

It's very poor shaky-cam quality, unrehearsed and unprofessional!  And the first 5-ish minutes just introduce my setup.  You can skip to 5:30 to see the start of the problem recreation.

So then I had a quick look inside the ELS 1.4 source to check out places where change direction is handled, and found one interesting spot in the core.h ISR method where "if the feed or direction changed, reset sync to avoid a big step" and thought AHA, this is likely where the problem is happening.  Well, I tried commending out that reset when the direction is changed, but it didn't work as expected.  Now when I change direction, it tries to move the servo too fast and causes a "too fast / reset" error.  Another short video of that FWIW -- oh, but I see now that it was accidentally truncated before you actually see the "too fast" error when I change direction: https://photos.app.goo.gl/gUQWFHanskAgDRoE8   

I think that I'm on the right basic track, but a proper fix is going to need some more work.  If James happens to read this and knows a solution, I'd love to hear it!  If not, I'll dig in further when I get a chance and report back if I'm able to resolve the issue.  

But I'm also wondering why nobody else seems to be running into this?  Maybe because y'all have thread dials and always thread using the same imperial or metric as your lead screw so that you never have to do the back-out-and-reverse?  Or maybe it's unique to my using this servo rather than a stepper.  I'm using the ISV57T-180 "stepper killer" integrated servo: https://www.omc-stepperonline.com/servo-motor-nema-23-integrated-easy-servo-motor-180w-3000rpm-0-6nm-84-98oz-in-20-50vdc-brushless-dc-servo-motor-isv57t-180 .

More news as it happens...

Greg

clough42, 2022-07-29:
The ELS is behaving correctly in your video.

Both marks were spinning in the same direction, in sync, until you stopped
the lathe at time code 6:44 in your video.  At this point, both marks were
pointing toward the front of the lathe.  You then reversed the ELS and
restarted the lathe, causing the marks to spin in opposite directions (i.e.
they were only pointing in the same direction twice during each revolution
of the spindle).  At 6:46, you stopped the lathe again, this time with one
mark pointing up, and one pointing down.  You switched the direction again
and restarted the lathe.  At this point, the spindle and leadscrew were
spinning in the same direction, at the same speed, but approximately 180
degrees out of phase, and this is exactly the state it was in when you
checked and discovered it was out of sync.  They're out of sync because
it's doing exactly what you asked it to do.  This is also exactly what
would happen if you switched directions on a mechanical gearbox with a
tumbler lever.

The ELS will properly retain synchronization when changing the direction of
the SPINDLE.  To cut a metric thread with an imperial leadscrew, or
vise-versa, make your first pass, then leave the half-nut engaged and run
the SPINDLE backwards to reverse the carriage back for another pass.  Do
not touch any of the buttons on the ELS.  As long as you leave the ELS
engaged, it will behave just like a mechanical gearbox, and the leadscrew
will stay in sync with the spindle.  (There is of course still backlash, so
you have to back the tool out when reversing back for another pass.)

If you change any settings on the ELS (mode, pitch, feed, direction) it
resets the synchronization internally and makes no guarantees with respect
to any previous position.  When you commented out this code, you discovered
why it does this.  The ELS continuously tracks the position of the encoder,
applies a gear ratio multiplier, and calculates where the leadscrew should
be so it can drive it to this position.  If you press the REV button, it
changes the sign of this multiplier, and suddenly has a new desired
position far, far in the opposite direction.  It must null out this value
and start tracking from zero, or the code will try to actually drive the
leadscrew to that far, far away position.

The "TOO FAST" message is a safety circuit breaker to prevent the servo
from getting behind and continuing to spin after you stop the lathe.  This
doesn't normally happen unless you spin the spindle too fast for the servo
to keep up, but it can also can serve as a safety measure, protecting the
operator against something totally unexpected, like someone disabling the
code that zeros out the accumulator when changing direction.

James

gregreid, 2022-07-29:
Thank you for that detailed explanation.  So I've been performing my "back out and reverse" procedure entirely WRONG all this time.  I need to leave the ELS in the same (forward) direction but use my late reverse to back up the tool to begin the next pass.  OK, got it!

Big problem though: on my lathe, switching the spindle from forward to reverse completely turns off the lathe (safety feature) and I have to power it back on again.  I have my servo and ELS wired to the same power relay as the lathe, so everything will turn off and have to re-boot each time I switch spindle directions ... which would no doubt make the ELS lose position as well.

So it seems that I'll need to change the source of the 120V input power to my ELS 5V "wall wart" and to my 36V servo supply so that they remain powered at all times.  (i.e. separate physical on/off switch for them.)   Dang, lots of tear out and rewiring ... and I have to unbolt and move my lathe to get access back there ... but if I must, I must.

Sure would be nice if switching the ELS forward/reverse could somehow maintain its leadscrew position.  This would be so much more useful to MY particular needs anyway.  But of course I don't expect any more miracles.

Thanks again James!  Great help.

clough42, 2022-07-29:
I understand the inconvenience, but maintaining synchronization when
changing direction isn't a physical possibility.  It's not that the
software doesn't support it; it's that it isn't a thing that can be done.
If you reverse the direction and turn the spindle, you're telling the screw
to turn out of sync from where it would be if it were in the forward
direction.

gregreid, 2022-07-29:
Again I really appreciate your time and expertise.  Meanwhile, I went to the garage to confirm that turning the chuck backwards manually to somewhat beyond (for backlash) the thread start does of course maintain synchronization.  So now I'm pondering whether to rewire the 120V inputs to the ELS and servo or to perhaps bypass the lathe switch's safety feature so that it maintains power during forward/reverse changes.  

I'm also studying the ELS source code some more, trying to better understand its various synch-related variables and methods.  It seems to me that the leadscrew's current position could be kept as an integer and maintained (with positive or negative increments according to the servo direction) no matter which way it's turning, never losing synch with the spindle movement.  That is, as long as the RPM goes to 0 between direction changes -- which your code already ensures.  

Your latest response indicates this isn't possible but I'm struggling to understand why.  I'm sure it'll come to me once I dig in a bit more.  Thank you again James, and enjoy your weekend!

gregreid, 2022-07-30:
You can truly find ANYTHING on the internet.  Of course someone else had also been through the aggravating issue of many Chinese lathes having the Fwd/Off/Rev safety feature, turning off lathe power through the direction change.  I'd removed my direction-change switch and was looking up a schematic for it when I stumbled across this thread:  https://www.hobby-machinist.com/threads/odd-behavior-in-mx-210v-forward-reverse-power-switches.96024/page-2  by a guy called "Tomzo" in California.  He figured out that simply connecting together the neutral lines that are originally wired into positions 9 and 10 of the rotary switch will keep the main on/off relay latched ON during a direction change.  

I just did the simple mod myself and it works perfectly.  Yay!  So now I can leave the input power to my 5V and 36V ELS and servo power supplies still connected to the lathe main power relay while being able to conveniently change spindle directions, maintaining ELS and servo power.  (Of course I'll need to make sure that the spindle is stopped before changing direction, now that the safety feature is bypassed.)

Oh, and if you read Tomzo's thread linked above, you'll find that he too is adding the Clough42 ELS.  Small world.  :-)

I've not been able to say "thanks" to Tomzo in that hobby machinist forum thread because I wasn't originally registered there.  I am registered now, but am awaiting a moderator to approve me before I can append my "thanks" to him ... which I'll certainly do.  I can also share my implementation and 3D printed parts with him (if he's not completed yet).

clough42, 2022-07-30:
In response to your previous comment about not understanding why it can't
stay in sync when reversed:

It can't stay in sync because staying in sync forward and staying in sync
reversed are two different things that require the screw to be in different
positions.  Consider these two scenarios.

1. You have the ELS in forward and turn the spindle clockwise ten degrees.
The leadscrew would follow and turn clockwise ten degrees.
2. You have the ELS in reverse and turn the spindle clockwise ten degrees.
The leadscrew would follow and turn COUNTERclockwise ten degrees.

These are two different positions, so it can't be in sync for forward
movement and backward movement at the same time.  So when you switch to
reverse and go back to the beginning of the thread and switch to forward
again, the screw is almost certainly in the wrong position and to
compensate, the ELS would have to make some kind of rapid movement to bring
it back into synchronization.  This seems like it might be practical,
because at most you would have to move it 1/2 turn, but it's actually more
complicated than this because in any real scenario that matters, there are
two additional factors: (a) you may be using the threading dial so not only
do you have to have the screw in the right orientation, but you have to
have the half nut locked on the correct thread.  The ELS would need
additional knowledge or would need to make additional assumptions to turn
the screw several revolutions to get to the correct position.  And (b)
metric and imperial thread pitches have a prime factor of 127 between them,
so the situation can become very complicated when trying to figure out
exactly where to move the screw and this necessarily involves making
assumptions about what the user intends.  The simple 1:1 case you tested is
easy.

Because this is complex, requires me to make assumptions about what the
user is doing, and because of the risks associated with unexpected movement
of the machine when pressing the mode buttons, I opted to emulate a
mechanical gearbox as closely as possible.  Given the broad appeal of the
ELS and the large variance in skill sets of the people who install it, I
personally think having something that moves automatically in potentially
unexpected ways is too big of a safety risk.  Imagine reversing back to the
beginning of your thread, advancing the cutter for the next pass, pressing
the REV button and having the ELS shift the screw, crashing the tool into
your workpiece, your live center, or some part of your body.

I think most advanced users who want more features like auto stop, auto
recall, automatic multi-pass cutting, and taper turning would be happier
with a CNC conversion.

James

gregreid, 2022-07-30:
Terrific explanation.  Thanks for taking the time to write all that.

Now that I've bypassed the "power off when changing spindle directions" safety feecher of my lathe, so that I'm able to Fwd/Rev the spindle quickly while maintaining ELS power and synch, I'm perfectly OK with using the traditional/proven/correct method that you described in your first response.  That is, leaving the ELS direction unchanged, and using the spindle Fwd/Rev to back out to start the next thread pass.  Works great, and I've cut some beautiful looking threads now.

If you're ever in the Daytona Beach area, several beers have your name on them here.

Chucksbp, 2022-07-30:
By the way, that's a very nice installation and a great video! And as always a great explanation by James!

51mone, 2022-09-05:
Maybe adding the option to use 1 of the readouts to display a virtual threading dial/index would help? Well, as long as it would be easy to program.

clough42, 2022-09-06:
It isn't that simple.  To create a virtual threading dial, the software
would also need a digital scale to read the position of the carriage.

James

clough42, 2022-10-11:
Excellent.  I'm glad you found a solution.