clough42/electronic-leadscrew #225: Custom Thread Pitches

RJSakowski, 2022-08-08:
A useful added function would be to be able to enter custom thread pitches from the user interface. I expect this could be done by temporarily redefining button functions.

SMaclund, 2022-08-08:
What's your use case for this that wouldn't be just as easy to permanently add it in the tables?

RJSakowski, 2022-08-08:
The reason for doing this would be non standard pitches which might be used for winding coils or springs.  The pitch for Close wound coils or springs is determined by wire diameter. There are also some vintage threads with odd pitches.

I had added a custom pitch to the tables in the past.  Each custom pitch done that way requires reflashing the Launchpad though. I seem to have seen that there is limited room in the tables and, in any event, adding all possible pitches for the possibility that one may be needed someday would clutter up the table and make scrolling through ti find one of the more common pitches more tedious.

kwackers, 2022-09-09:
If it's any help I've just changed the code to allow custom pitches.

These are accessed by using the 'Set' button and selecting the (only) option 'Cust Thd'.
You can then change the 3 values for thread pitch by using the up and down arrows and Set button, the final press of Set will allow return to normal operation but using the custom thread.

It only supports metric - any imperial threads will require a calculator.

Pressing the Feed/Thr or In/mm buttons will return to the normal threading tables, however you can access the custom thread again via the Set button menu (value is retained until powered off).

kwackers, 2022-09-15:
I've created a fork with the custom pitch code in here:
https://github.com/kwackers/electronic-leadscrew