clough42/electronic-leadscrew #227: Make is missing

lesrhorer, 2022-09-08:
As fsar as I know, the IDE install should have included the make executable or at least made sure it existed and was in the path.  It does not exist, so I cannot build the code and install it to the MCU.

pmelevende, 2022-10-04:
Hi Les - Open the IDE, Open the project in the project explorer, Select 'els-f280049c' as active project, then click on the little hammer icon. That should build the project for you.

rak306, 2022-10-14:
I'm having the same problem. Error - no make file. also warning - no include path. downloaded latest version of CCstudio (version 12 on 2022-10-14). Downloaded electronic-leadscrew1.4 from github.

in CCstudio File-Open projects-the electronic v1.4.
in CCstudio click on els-f280049c - it goes bold
click on hammer - 1 error - no make file, nothing to build
also warning - no include path

tomvikse, 2022-10-14:
NOTE: The code for this project was developed in Code Composer Studio version 9.1. It has not yet been migrated to version 10 or later

rak306, 2022-10-14:
And TI only has version 12 of CCStudio available on their website. So if the project doesn’t compile on the version 12, the project is not available to newcomers.

Now I was able to find CCStudio 9 by using the wayback machine. And that worked. But it was difficult because the download kept failing after a few hundred megabytes.

tomvikse, 2022-10-14:
see my link..you can download v.9 there..

clough42, 2022-10-16:
For people coming along later and finding this thread click "View All
Versions" on the TI CCS download page, and you'll see a listing for
previous versions, all the way back to version 4.

Porting the project to newer versions of CCS is something that's on my list
of things to do, but since it doesn't actually add new value to the ELS
project, it has not come to the top of my list.

James

pmelevende, 2022-10-17:
Hi James,

Maybe then, it's close to time to 'add some value' :-)

>

pmelevende, 2022-10-19:
Hi again,

It's been a while but something was nagging me in the back of my mind that I'd updated and got it to work - So - I checked the version of CCS I'm running - It's 10.4.0 and it successfully compiles and loads the project. Wish I could remember how I made it work!

All I can say is that 10.4.0 works for me.

Thanks
Pauly