clough42/electronic-leadscrew #228: Issues with ISV57T-180

caskscob, 2022-10-12:
I have the servo wired up and it runs fine in either direction.  Not sure of the speed yet because I haven't been able to get past the power issue.   The issue is that when I engage my cross slide the motor stalls.   I have a 20T on the motor, 60T on the drive.   My lathe has a gearbox where I can select 1:1, 2:1, and 4:1.   If I engage at 4:1 it works fine, 2 or 1 to 1 and it will stall.  

I have the dip switches set on the motor for 2000 steps/rev. 

In the code I put this: 
#define STEPPER_MICROSTEPS 3      
#define STEPPER_RESOLUTION 2000 

Any help would be greatly appreciated.  

Wiring shown below.  
[image]
[image]

caskscob, 2022-10-13:
I have changed the microsteps to 16.  I did that because when I tested the unit on the bench it worked great and that is what I had in there.   It now has a lot more power but still not enough and the speed is incorrect for threading.  Does someone know what the values are I need to put in here to make it work properly?    Thanks

Chucksbp, 2022-10-14:
Sounds as though you don't have sufficient motor power what motor did you use and how large is your lathe? 
 the 2000 resolution and 3 micro steps should work with the 3 to 1 pulley ratio and a 1 to1 gear box ratio. Im not positive but I believe the micro steps would need to be 12 for the 4 to1 gear box ratio???

caskscob, 2022-10-14:
The motor is the ISV57T-180 as shown in James' video.    When I set the steps to 2000 and 3 the motor doesn't turn at all even with the lathe out of gear.  I have a G0949  12 X 36 lathe.  That ISV57T-180 motor is a beast really so I don't believe it is the motor not being powerful enough.  I need someone who uses this to tell me what the settings need to be I think.   I would think that with the 3:1 pulleys and with a 2:1 gearbox setting, I would have 6:1 total.  Not sure what the settings should be for microsteps!

Chucksbp, 2022-10-14:
From my understanding of the steps and micro steps The steps are what you have the motor set at dip switch related. the micro steps seem to be the final gear ratio the motor will see. so in my opinion you are correct at 6 micro steps 3 to 1 pulley ratio times 2 to 1 gear box ratio would be 6 when i went to school. LOL! are you getting proper voltage to the motor??? my thinking is the motor is the problem. the motor should be almost unstoppable even if you are turning the chuck by hand if the motor is correct. My understanding the electronics only drive the motor driver the motor driver tells the motor through micro steps what to do. lack of power isn't the fault of the Clough42 part???? You may want to contact the people that supplied the motor and controller. This was my first attempt at this type of setup although I have some electronic back ground from 53 years ago. Shortly after my build i lost tach signal and motor operation all together it turned out to be am encoder failure that wiped out the number one TI board input. a new encoder and re flash to port two fixed the issue. 
   Just for reference i have my motor running at 1,000 resolution with a 3 to 1 belt ratio and no gear box, so i am at 1,000 resolution and 3 micro steps and threading has been prefect. I hope you get it worked out, Its a great system!!!!! you will love it!!! its worth the effort. good luck.

caskscob, 2022-10-14:
chuck,  Thanks for your reply!   I will give it a shot at setting microsteps to 6 now.  I know 3 doesn't work at all for whatever reason.   I wonder if an odd number doesn't work on the motor.

tomvikse, 2022-10-14:
for 1:6 ration:

Connet on to the servo and set PR 0.08 to 500..
https://ibb.co/xhHrKNP

In Configuration.h 
#define STEPPER_MICROSTEPS 6
#define STEPPER_RESOLUTION 500

caskscob, 2022-10-17:
Tom, 
I changed PR0.08 to 500.  I set the microsteps in the code to 6.   when the control panel is set below .010 it is still too weak to engage the sled.  Above .020 you can engage the sled but if you put a little pressure on it the motor stalls.   How did you come up with those settings? are you using them?    Thanks.

tomvikse, 2022-10-17:
Spec servo power supply? I use a 48v 350w 
Yes use this setting. 
my setup 
iSV57T-180
10 on servo 60 on leadscrew 1:6
Lathe
https://www.holzmann-maschinen.at/EN/metal-lathe-552

caskscob, 2022-10-17:
Hmmm.   I can't find a 10T pulley with 8mm shaft in a 15mm wide belt.  I may switch to a 10mm belt.   I'll connect 48v to it to see if that matters first.    I have a variable 1500W supply I can connect to it for test.

Chucksbp, 2022-10-17:
Check this place out for your pulley you need. good chance they may have something. I had to boar a couple of mine for the correct shaft size. For my lathe i needed a 3/4 with a key. in a 60 tooth. 
https://sdp-si.com/tools/center-distance-designer.php

caskscob, 2022-10-19:
Ok.  Thanks all for your help!!  I ended up just experimenting with the steps to get it right.   Turns out that microsteps at 6 and steps at 1000 works spot on with the gearbox set at 2:1 using the 20T and 60T pulleys.  I'm not going to try figuring that out.  
I ran it at 48v to get it working and cut some threads with it.  I then tried it at 36V with the original supply as it is mounted in the lathe nicely and is quite small.  Works perfect.  You can use a 36V-180W supply without issues.  I did some test cuts in steel up to .075 depth at .010/rev and it didn't blink, doubt I'll need more than that.

closed 2022-10-19 by caskscob