clough42/electronic-leadscrew #229: No RPM readout

jojotoshio, 2022-10-13:
Hello,

I am setting up a new build. Currently have completed programming the TI board and have assembled the interface and control panel. It appears to power up as expected. The LCD screen works and the buttons adjust the numbers properly. However, when attaching the encoder, I do not get a RPM readout. I am using an omron  E6B2-CWZ6C. I have checked the continuity of all the connections and wires. I even tried switching to the EQEP2 header in the code but still no RPM readout. When checking the voltage where the encoder connects to the TI board, I get 5v at the 5v pin, 1.10v at 1A and 1.20 at 1B. Any ideas on what to try next? Thanks!

caskscob, 2022-10-13:
Is your encoder wired like this?
[image]

jojotoshio, 2022-10-13:
> Is your encoder wired like this? [image]

Hi, yes the encoder wires are attached like in the picture. Thank you for your help.

caskscob, 2022-10-13:
Do the function buttons on the display control work?

jojotoshio, 2022-10-13:
All the buttons function properly on the display. Do you know what voltage at 1a, 1b or 2a, 2b should be?

Chucksbp, 2022-10-14:
Is there no display only?? does the motor function???? If not I would try reprograming the TI board for the number 2 encoder input and probably try a second encoder. some of the encoders are bad, they are cheap Chinesium knock offs.

jojotoshio, 2022-10-14:
> 

Display works. Servo motor is not attached.  Also tried reprogramming for the  EQEP2 header with no change. Unlike the problems discussed in #197, my RPMs never displayed, it only reads 0.

jojotoshio, 2022-10-17:
Issue is now resolved. Turned out to be a bad encoder. Thanks!

closed 2022-10-17 by jojotoshio

Chucksbp, 2022-10-17:
Glad you got it operating! the encoders seem to be the problem most of the time! what can you expect for the price point!!!! 
once you get it running you will love the system. I did some single point threading today 3/8 24 took no time hit a couple buttons dropped into back gears and away it went done in a few minutes, easier than a hand die.

chaszz, 2022-10-27:
I had a similar issue. Running great for a year and then the lead screw just stopped. The RPM shows zero even though the encoder is turning. I had a spare encoder and plugged it in and gave it a spin. Still no RPM so it must not be a bad encoder. I read the #197 discussion and saw that there is an option to use an alternate encoder input in the configuration.h file. Sure enough, I looked on the TI Board and there is another set of input pins immediately to the right of the input pins my encoder was connected to. I changed the configuration.h file to use the alternate input (EQEP2) and switched my connector to this input. Lead screw is now operating fine. I have ordered a backup TI F280049C board.

Chucksbp, 2022-10-27:
I wonder if there wasn't a bad batch of boards??? Not that I am bad mothing TI! With all the chip and component issues there may have been some issues. Again its hard to really say. It would be interesting to know how many previous failures there were. For the cost it's defiantly worth while having some spare parts for this great tool! I would be lost now that I'm so used to having it!

chaszz, 2022-10-27:
Agreed. I don’t do threading everyday, but just the ability to change leadscrew speed on the fly without messing with change gears is wonderful.

Chucksbp, 2022-10-27:
My lathe is an old Atlas 618, strictly change gears. Years ago I converted my leadscrew to a 12 volt motor with a variable power supply that i used for turning only. I would remove one gear so that it could free wheel. It worked OK but I didn't have fine control over the speed. then if I wanted to thread i had to take the belt off and set up the change gears. so most of the time i did my best to thread with a die which isn't the easiest to do. 
   The Clough42 lead screw brought my probably 80 year old lathe into the 21 century, especially with the addition of a DRO! I'm retired and do some little project almost every day during the winter.