clough42/electronic-leadscrew #231: Thread to shoulder

kwackers, 2022-10-16:
I've been working on thread to shoulder code - I can't see anywhere else to put this so I've put it here as an "issue".
This is still very much work in progress but superficially works on my desk and did work on my lathe last time I tried it.

If you're not risk averse then give it a go and feel free to add feedback.
See appropriate branch at [https://github.com/kwackers/electronic-leadscrew](url)

Some notes.
Currently config is "my" config - so you'll need to change anything necessary in there. There are also a number of other changes to the way it works (default settings on power up, number formats in metric etc).

I've added a menu system to the 'Set' button, this cycles through 3 options currently available "Custom Thread", "Shoulder" "Position".
Press 'Set' again at any point to enter that mode, when in a mode the first press of the power button exits. 
If no mode is selected then the menu system will time out after a few seconds and return to the normal screen.

Custom thread use is covered in the updated details file.
Position simply shows the angular position of the spindle (I borrowed the idea from another fork whose name I've forgotten - apologies).

So to use thread to shoulder, set up your pitch, direction, in/mm etc as per usual.

Press 'Set' to get the menu up and use the up/down buttons to cycle the options then at "Shoulder" press the 'Set' button again.

The message "Go Shldr" appears, move the carriage to the shoulder and if you haven't already engage the half nuts (at this point the half nuts MUST remain engaged until you've finished cutting your thread and exited the mode).
Press 'Set' to mark the shoulder position.

"Go Start" now appears, you now move the carriage to the start of the thread by spinning the spindle, once there press 'Set'.

(That's the setup completed. You've set the thread and marked the shoulder and start positions)

"Begin" now appears momentarily. At this point you can cut the thread as usual.
When it reaches the shoulder the carriage will stop moving and the word "Stop" will appear, you should now stop the spindle.

Once the spindle has stopped the word "Retract" will be shown, retract the tool and then press 'Set' the carriage will now return to the start position and "Begin" will once again be momentarily shown.
(note that the carriage can move past the start by up to the pitch of the leadscrew in order to re-establish sync with the spindle. 
You can also rewind the carriage the old way by spinning the spindle - not sure why you would but it will work.)

Simply repeat the above, when finished press the power button once to exit shoulder mode.

Some things to note.
It's only had minimal testing on my lathe so there may well be bugs.
I need to lock out buttons - changing pitch whilst in shoulder mode will likely break things.
Some form of indication it's in shoulder mode (or indeed any mode is needed).
Still have some tidying up to do in the code, particularly to try and make the style consistent with the surrounding code.
Probably lots of other stuff I haven't thought of yet.

As I said this is work in progress but I think it's at a point were anyone interested who wants to give it a go can do.
In the meantime I'll do some testing, make some changes and once I'm happy it's where I want it to be I'll push the changes up.
Once that's done I'll add multi-start threads (since a lot of the work for them is already done.

(P.S shameless self plug but if you're interested in automating things have a look at my website www.worldofward.com)

tomvikse, 2022-10-16:
Testing it out now but only get custom thred in the menu..

kwackers, 2022-10-16:
> 
Have you picked the correct branch? (You want the "thread to shoulder" branch and not master).

Also probably should have made it clearer you have to use the "+" and "-" buttons to step through the menu.

tomvikse, 2022-10-16:
Did not see the branches, github is new to me..

tomvikse, 2022-10-16:
first test,stoped fine, on the way back to start it stoped 1/2 way and got a error "reset too fast"

tomvikse, 2022-10-16:
There is something wrong whit the fees speeds to.. way to fast.. threading speed is ok.. testet 1mm pitch 10mm long..

And can you move the tretract button to the + button,way to easy now to hit the power button in stead of the set button and you reset the program..

kwackers, 2022-10-16:
> first test,stoped fine, on the way back to start it stoped 1/2 way (around 33mm) and got a error "reset too fast"
> 
> https://photos.app.goo.gl/hua9LtjjX59gnH2H9
> 
> sounds like the leadscrew servo try to speed up like crasy and trips..

That's a bit weird.
The error is basically to signal that the carriage position is too far away from where it should be.
During retraction this is actually true since the spindle isn't rotating but it's flagged so the error can't occur.
For it to occur the retract must exit early before it's reached the start (which is what looks like happens), the servo speeding up would happen at that point because it's now trying to catch up.

Can you attach/send me your configuration file? (I'm wondering if any numbers are overflowing).

The feed rates are set for my machine so you probably want to put them back to what they should be.
(When I remerge my changings back into my master branch I'll reset all of these to the defaults that were set in the original).

tomvikse, 2022-10-16:
https://www.filemail.com/d/kdenckvabxeyuwv

kwackers, 2022-10-16:
> electronic-leadscrew-ThreadToShoulder.rar https://www.filemail.com/d/kdenckvabxeyuwv

Cheers.
I have a repro. Hopefully won't take me too long to work out what's happening.

kwackers, 2022-10-16:
New version pushed up with fix.

kwackers, 2022-10-24:
Added multistart threads

tomvikse, 2022-12-09:
There is a big in there.. se post i made on your fork

steelbarz, 2023-02-19:
Have been using the Clough42 ELS for over a year and love it,  This feature is exactly what it needed.  Just loaded and updated and seems to work great thank you very much (PS Love your rotary table upgrade too been using it as well)

blackfister, 2023-03-31:
I have installed your version, and doesn’t seem to work, no action with + - , weird. While inhaling cloughs els version again, I lost threading, it advances, but in thread mode, stepper doesn’t activate, any hint on this? Thanks in advance

kwackers, 2023-03-31:
Hi, sorry I can't offer any hints but reverting the firmware should work - there's absolutely no reason why it wouldn't.
Sounds like something else has broken - something popped off the board? Cable broken?
Once you have the Clough version working correctly then you should be able to install mine - again it should work just fine (assuming you copy across any settings you need).

> I have installed your version, and doesn’t seem to work, no action with + - , weird. While inhaling cloughs els version again, I lost threading, it advances, but in thread mode, stepper doesn’t activate, any hint on this? Thanks in advance