clough42/electronic-leadscrew #233: Servo gearing & micro-stepping

MorryStu, 2022-10-28:
I have a fairly simple question

Due to how I can mount the servo, I want to run gears instead of belts, and use the gears that came with the lathe to drive the leadscrew

On the servo the smallest gear I can run is a 40t as any smaller and the gear dia is inside the external dimensions of the motor, and I want to direct gear it

The biggest leadscrew gear is 60t, which I can get to mesh with the 40t very nicely without resorting to making/hacking up some sort of oddball bracket, ( and the lathe end cover) it is also about as close as I can get to meshing properly with a simple sturdy mount

The question is, that in config.h, will it accept a decimal number or whole number only as 60t/40t = 1.5

so will the following work

// Steps and microsteps
#define STEPPER_MICROSTEPS 1.5
#define STEPPER_RESOLUTION 1000

or, would this work

// Steps and microsteps
#define STEPPER_MICROSTEPS 3
#define STEPPER_RESOLUTION 2000

and yes, I could just go buy another 60t, but I dont want to if I dont have to

tomvikse, 2022-10-28:
Is the servo strong enough to run 40/60 gearing?

MorryStu, 2022-10-28:
Its a 4.0Nm. But If I need a bigger one then I will get one

My question is about stepper settings, not the strength of the motor. It could be a 2Kw servo motor, same problem

kwackers, 2022-10-28:
The second will work just fine.
I use some really oddball numbers to make my feed work (it has multiple bit of gearing compared to the leadscrew).
I got them empirically by rotating the spindle 10x and noting how far it had moved.

I ended up with.
#define STEPPER_MICROSTEPS_FEED 16
#define STEPPER_RESOLUTION_FEED 1157

Chucksbp, 2022-10-28:
I would say that if your motor is set at 1000 stepper resolution and you use the 1000 stepper resolution setting in the software along with the1.5 Micro steps along with the proper setting for the lead screw pitch you should be fine.  the lead screw doesn't care how its driven. You may have to set the direction, that will depend on how your motor is placed. but that is an easy task. you may be able to change that at the motor control with a switch?? or it can be done in the software.