clough42/electronic-leadscrew #234: Connection for Nema 23 Intergrated servo to Macbook pro

stuwindsurf, 2022-10-29:
Has anyone managed to connect the servo James recommends - Nema 23 Intergrated servo - to a Macbook pro?.....

I have the VGA lead that came with the motor but my Macbook only has C type inputs. I have seen adapter cables to convert VGA to C type but they suggest they are for video converting? Does anyone know if they would work for this application?

This is the one I am looking at getting - https://www.amazon.co.uk/gp/product/B07HF9BQJD/ref=ox_sc_act_title_1?smid=A2XXZWEC7I5R18&psc=1

Any help would be much appreciated. Also I have managed to download the CCStudio on my Mac and dropped the latest ELS version 1.4.00 to change some of the code for my application, but I am unsure how to flash it on to the TI circuit board as the CCstudio doesn't seem to have the hammer icon James mentions or the bug icon!!

As you've probably guessed I am a really dummy when it comes to computer coding.

Thanks
Stu

JonWoellhaf, 2022-10-29:
Hi, Stu.

My thought is that you are headed down a deep rabbit hole if you continue to use a Mac for this project. I recommend you get a cheap Windows machine.

Good luck!

Jan (Yan)

From: stuwindsurf 
Sent: October 29, 2022 08:39
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: [clough42/electronic-leadscrew] Connection for Nema 23 Intergrated servo to Macbook pro (Issue #234)

Has anyone managed to connect the servo James recommends - Nema 23 Intergrated servo - to a Macbook pro?.....

I have the VGA lead that came with the motor but my Macbook only has C type inputs. I have seen adapter cables to convert VGA to C type but they suggest they are for video converting? Does anyone know if they would work for this application?

This is the one I am looking at getting - https://www.amazon.co.uk/gp/product/B07HF9BQJD/ref=ox_sc_act_title_1?smid=A2XXZWEC7I5R18&psc=1

Any help would be much appreciated. Also I have managed to download the CCStudio on my Mac and dropped the latest ELS version 1.4.00 to change some of the code for my application, but I am unsure how to flash it on to the TI circuit board as the CCstudio doesn't seem to have the hammer icon James mentions or the bug icon!!

As you've probably guessed I am a really dummy when it comes to computer coding.

Thanks
Stu

tomvikse, 2022-10-29:
From Wiki..

NOTE: The code for this project was developed in Code Composer Studio version 9.1. It has not yet been migrated to version 10 or later

https://www.ti.com/tool/download/CCSTUDIO/9.3.0.00012

stuwindsurf, 2022-10-30:
> https://www.ti.com/tool/download/CCSTUDIO/9.3.0.00012

Thanks, I will give that a try

stuwindsurf, 2022-10-30:
> 

Mmmm..... I think you may be right!

tomvikse, 2022-10-30:
use this one and just get a USB hub to get from usb c 
https://www.omc-stepperonline.com/rs232-adapter-cable-to-usb-2-0-c003

stuwindsurf, 2022-10-30:
> use this one and just get a USB hub to get from usb c https://www.omc-stepperonline.com/rs232-adapter-cable-to-usb-2-0-c003

Thanks, is that different in terms of use/compatibility 
compared to the one in my link above?...

stuwindsurf, 2022-10-30:
...this one 
https://www.amazon.co.uk/gp/product/B07HF9BQJD/ref=ox_sc_act_title_1?smid=A2XXZWEC7I5R18&psc=1

tomvikse, 2022-10-30:
you need a RS232 Adapter not a VGA cable..

stuwindsurf, 2022-10-30:
> you need a RS232 Adapter not a VGA cable.. the RS232 Adapter has 9 pins

Thank you very much for your help

closed 2022-10-31 by stuwindsurf