clough42/electronic-leadscrew #235: How to flash code to TI launch pad

stuwindsurf, 2022-11-01:
Hi,

I'm having trouble figuring out how to flash my configured code to my TI launch pad! I notice in James video he selects a USB filter? which seems to enable him to flash the code direct from CC studio.

I am running CC studio in Windows on my Macbook so don't seem to have that option. Can anyone help explaining another way I can flash across to the TI board?

I am a real dummy when it comes to computer programming so can only follow instructions rather than figure stuff out.

Many thanks 
Stu

Chucksbp, 2022-11-01:
I did mine on my home computer with windows 10 and used no filters. Make sure you download the latest version 9 of the cc studio.  after configuring I just pugged the supplied cable into the launch board and computer. I did not use an external power supply i just left the jumpers on the TI board. I downloaded the software as per video, took less than a minute. Then i removed the three jumpers installed the Clough42 board to the TI board and installed everything into my electronic box. I HOPE THIS HELPS.  I also struggle with this type of computer work, but honestly the biggest problem I had was getting the program downloaded. I had the video running and on pause and just followed everything step by step.

stuwindsurf, 2022-11-01:
...exactly what is the supplied cable for plugging the TI launch board in to the computer? I have been using a standard USB cable, is there a specific cable required for this?

Thanks 
Stu

Chucksbp, 2022-11-01:
Both of my TI Boards came with a short Micro to USB cable, I'm sure its nothing special although some cables that come with phone chargers are only for charging and wont transmit data. They don't have all the wires just the plus and minus 5 volts. So i figured they sent Ill use it. most cables you buy will be for data.

tomvikse, 2022-11-01:
> Thanks. I did try version the latest version of CCS but it didn't have the same icons as the version 9 James uses in his video, so I downloaded version 9 and have managed to upload and configure the els software for my hardware. The problem I have is flashing it on to the TI launch pad. As mentioned I am using a virtual version of Windows on my macbook and don't have the USB filter option James uses for automatic connection when plugging in the TI board. So the only thing I can think of using is a Flash program, but I have no ideawhat to use and how to use it! ....

there is no "filter" maby you mix autodetect of COM port and filter.. did you try "Mac OS single file installer for CCS IDE V9"?

tomvikse, 2022-11-01:
Jut use Mac OS single file installer for CCS IDE  v.9

Chucksbp, 2022-11-01:
No I have a windows 10 system so everything works as it should. I didn't use any filter. I actually programed my first board twice because the number one encoder input failed. Then I programed a second board for a spare while I remembered how to do it. I tested it and it was fine also. Sent from my Galaxy

  pad (Issue #235)

tomvikse, 2022-11-01:
my bad.. stuwindsurf uses VM so need some sort of filter to get USB to work.. but easyer just to use the mac version 9 and not VM

stuwindsurf, 2022-11-01:
> Jut use Mac OS single file installer for CCS IDE v.9 https://dr-download.ti.com/software-development/ide-configuration-compiler-or-debugger/MD-J1VdearkvK/9.3.0.00012/ccs_installer_osx_9.3.0.00012.dmg

Yes I did try that first, here is is a screenshot of it open. Where would I go to flash a file to usb?...
[image]

stuwindsurf, 2022-11-01:
...also I get this message when I try to open the file in workspace
[image]

Chucksbp, 2022-11-01:
I wish i could help you more. It appears as though your work hasn't migrated to the program. There is no data there at all. there's nothing to program. you should be able to program the chip right from the screen that you put your configuration H information in.

Chucksbp, 2022-11-01:
From what I remember, you should be able to program the board right from this screen Push the arrow and then when it stops hit the continue button then its finished. 
[image]

stuwindsurf, 2022-11-01:
Success!!! 😃😃  I went back to the CCstudio on the virtual windows setup on my Macbook, plugged the USB cable in again, selected release then clicked on the bug icon ....and.......this time it erased the existing data and flashed a cross the newly configured data!  I have no clue as to why it wouldn't work before but this time it did and there was an audible note when I plugged in the USB lead connected to the TI launch pad.
The mac version of CCS was dead in the water but the VM windows version seems to have worked!! Yaay

I just hope I can repeat the operation successfully again if I need to mess about more with the configuration settings. But for now I am one happy relieved chap!!

Thanks for everyone's input 👍

stuwindsurf, 2022-11-01:
> From what I remember, you should be able to program the board right from this screen Push the arrow and then when it stops hit the continue button then its finished. [image]

Thanks for all your help Chucksbp

closed 2022-11-06 by stuwindsurf