clough42/electronic-leadscrew #236: Problem comminicating with Nema 23 integrated servo

stuwindsurf, 2022-11-06:
Ok, so after all the problems I had with configuring the TI launch pad,  I now can't seem to find/communicate with the Nema 23 integrated servo that James recommended. I have the correct leads supplied by Stepperonline and downloaded the software from their product page but when I connect the leads and click connect or search it can't find the servo?? I get a message 'connect to drive fail,maybe abnormal or port be occupied'

As mentioned in my last post I am running Windows 10 through boot camp on my macbook. I also tried on my father's PC also running windows 10 but neither will detect the motor. I have emailed stepperonlines tech support and waiting for a reply. But in the meantime I wonder if anyone has encountered the same issues or have any advice?

I must admit I have come close to uninstalling all the hardware I mounted and resorting back to the mechanical gears. I have found the whole setup/configuration of the whole system very frustrating. 

Thanks
Stu

Turner75, 2022-11-06:
Hi Stu
Stick with it, the results are well worth the effort, and think of it as a
learning experience.
Cheers Graham

stuwindsurf, 2022-11-06:
Thanks Graham, I appreciate your reply and encouragement,  but I have not been able to use my lathe for the last 2 weeks trying to configure everything and unless I can communicate with the servo I'm not going to be able to complete and use the Clough42 ELS conversion.

It seems the easy part is fitting the hardware, I actually enjoyed that part and am happy with the setup. But it's no good to me if I can't get it configured!!  Maybe Stepperonline will come back with a solution!

Chucksbp, 2022-11-06:
why do you have to communicate with the servo??? doesn't it have dip switches on the controller to set it up. The Amazon one i got I was able to set up with switches right on the controller no other adjustments was necessary.

stuwindsurf, 2022-11-06:
Yes it does, but the lowest resolution I can set with the dip switches is 1600 and I want to set it to 1000 so I can set the microsteps to 4 for my 4:1 drive ratio on the leadscrew

Chucksbp, 2022-11-06:
You can set the resolution to 1600 in code composer to match your motor controller setting and still use the 4 for micro steps for your gear ratio.  Those are  is two different settings.

stuwindsurf, 2022-11-06:
Thanks, I realise that but not knowing anything about resolution or microsteps I am just trying to follow James setup. He mentions in his video about the Nema 23 servo (Is it a clearpath killer) that he wants to set the resolutionto 1000 or 500.so I figure there is a reason for that?...

stuwindsurf, 2022-11-06:
....also in the config menu for the ElS I think it said if you have a leadscrew drive ratio of 3:1 set the resolution to 1000 and the microsteps to 3, hence I want to set the servo res to 1000

Chucksbp, 2022-11-06:
The only problem I could see is possibly he doesn't have that large of a number in his software??? I tried it in my code compressor and it didn't throw an error code. All you can do is try it and see if it will install it on the board. Unless you find a way to load the motor control software. Does the controller need powered up to make changes? or does it just go through the network cable? im not familiar with changing motor settings that way.

Chucksbp, 2022-11-06:
This is from the description in setting up the motor in James description. . You need to define the number of steps and microsteps for your stepper or servo motor, so the firmware can calculate the correct drive ratios. This is defined as steps and microsteps. A typical stepper motor has a resolution of 200 steps per revolution, and a typical driver configuration would be 8 microsteps per step. This means that the driver will require 1600 total step pulses (8 * 200) to turn the motor one revolution. If the drive ratio between your motor and leadscrew is not 1:1, use the microsteps value to compensate. For example, if you have a 1000-step servo (common) and your drive ratio is 3:1, set STEPPER_RESOLUTION to 1000 and STEPPER_MICROSTEPS to 3. As long as the multiple of these two values is the number of pulses it takes to rotate the leadscrew one revolution, it should work.

stuwindsurf, 2022-11-06:
I have tried communicating with the motor with it both powered and unpowered but neither worked. I believe it just needs 5v from the USB port for the controller to be able to connect to the motor control software

stuwindsurf, 2022-11-06:
> 

That's the part I can't seem to get my head round and is why I am trying to set the servo res to 1000 then I know I need to set the microsteps to 4 for my 4:1 gear ratio

stuwindsurf, 2022-11-06:
... that's presuming the Nema 23 servo has 1000 steps?? I can't seem to find that information

Chucksbp, 2022-11-06:
From what I read, James software allows up to 2000 for stepper resolution, so you should have no problem operating at 1600 stepper resolution and 4 micro steps to adjust for your gear ratio.

stuwindsurf, 2022-11-06:
Ok thanks, I will give it a go later. I do appreciate you looking at this for me.

Chucksbp, 2022-11-06:
I think the 1000 number keeps popping up because many motors seem to be preset to the number when received. not all but many.  I used this motor and driver, it was one that James tested and sort of approved, it came set at 1000 so I just keft it there and set my micro steps to 3 for my 3 to 1 ratio. I helped another fellow with a similar issue that's where we found that 2000 was the max in the software. This is the motor I used, I had some gift cards and I am an Amazon and ebay buyer. i like that they bothe easily take care of any problem that comes along. https://www.amazon.com/gp/product/B079CCPCPQ/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1

stuwindsurf, 2022-11-07:
Ok, quick update! Stepperonline got back to me and told me I need to power up the controller to be able to connect. I thought I had tried that but anyway I plugged in the 48v power supply and found when I selected port 3 in the software it found the servo motor!!! The only thing I can think of was I hadn't tried port 3 last time I had the power connected.

Anyway lesson learned after a lot of wasted time, if you want to connect your Nema 23 integrated servo to the software, you need to connect the power supply to the motor and then in my case select port 3 to connect (not sure if it is always port 3 for this particular motor). I wish these instructions were available with the motor when ordered from the supplier. 🙄

I hope this hopes someone else and saves them a lot of frustration and wasted time!

Chucksbp thank you for your help, as I said before it has been much appreciated 😊 Fingers crossed I can get the resolution/microsteps to work for my gear ratio.

stuwindsurf, 2022-11-07:
Finally here's some information I wish I knew before I started the software configuration as it would have saved me a heap of time and grief....
Firstly you can run the code composer software to configure the TI launch pad on a macbook using Windows boot camp, hold down the option key when starting up the macbook and choose windows .
When you download the software for the ELS from clough42 github page, drop it in to the 'workspace_v9' directory which is created when you set up the composer studio, I was trying to drop the file directly in to the studio which doesn't work. James actually mentions this in one of his videos but as there was a lot for me to take in, I missed it.
You can download the Nema 23 Integrated servo software on to the mac boot camp windows version 10 and it will work. When trying to connect the motor to the macbook, use the serial port cable that comes with the motor and also buy the RS232 to usb lead they recommend for that motor in the link. 
Make sure you have power to the motor, 48v in my case, then select COM 3 in the communication section of the software, this one worked for me as mentioned above.

I hope this helps someone else trying to use a macbook to set up all the software

Chucksbp, 2022-11-07:
Once you have everything up and running you will see it was well worth the trouble!