clough42/electronic-leadscrew #237: 1:1 ratio configuration

blackfister, 2022-11-15:
Hi all, I’m running everything in a 1:1 ratio.
Stepper nema 32.
I have an imperial 8 tpi lead screw.
Threads come off. 
any tip on configuration a per 1:1 ratio?
Spindle / encoder 1:1
Stepper/ leadscrew 1:1
Thank you for any time you could spend on this.
Kind regards

Chucksbp, 2022-11-15:
You will set the lead screw setting to 8 TPI. You don't mention what your encoder resolution is, but the setting is 4 times the resolution so if you are using the recommended resolution 1024p/r you set the encoder to 4096. you then set stepper micro steps to 1 for your 1 to 1 ratio and then set stepper resolution the to match the stepper resolution of the motor. usually accomplished with dip switches. it can be set up to 2,000 but if the motor can be set at 1,000 i would go with 1,000 
 make sure you use a 9 version of Code Composer.

blackfister, 2022-11-16:
Thanks, I’ll give it a try

blackfister, 2022-11-16:
Still won’t thread as desired…. I’ll post pictures of config