clough42/electronic-leadscrew #238: Can't input decimals in micro steps

dawesy72, 2022-11-16:
By nature of how I've had to set up my gear train I've got a 12t pulley on the stepper and a 39t on the lead screw drive. This gives me a 3.25:1 ratio. When I enter 3.25 in the micro steps it causes errors when compiling. Putting 3 in works ok but obviously causes in accurate threads/feeds. Any help appreciated.

clough42, 2022-11-17:
The steps and microsteps are multiplied together, and that's the number
that's ultimately used, so you can change both so they multiply to what you
want.  So if your steps are 200, for example, you could just set steps to
650 and micro steps to 1.  If you're also using 8:1 microstepping, you
could set steps to 650 and micro steps to 8.  Any combination that yields
whole numbers should work.

Only whole numbers are allowed.

James

dawesy72, 2022-11-17:
Thank you James for helping me out. My steps are 1000 so I'll figure it out. 
Thanks again.

Chucksbp, 2022-11-17:
When you figure your steps keep in mind that I believe the steps are limited to 2000. I could be wrong about that, I believe that came up in a discussion, but if you get an error you may have to divide by two micro steps to get into that sweat spot.

dawesy72, 2022-12-28:
sorted this now thanks, yep steps limited to 2000

closed 2022-12-28 by dawesy72