clough42/electronic-leadscrew #239: Resolving encoder not working with 5 volt supply

dlmock99, 2022-11-30:
Found that my encoder would work at supply voltages higher than 5 volts but not with 5 volts.  The problem is with the internal voltage regulator in the encoder a LM78M05.  Quick fix is to remove the regulator and solder a jumper from the input and output board pads.  The fix is good only if you are using 5 volts to power the encoder.

kwackers, 2022-12-01:
Is it a genuine Omron or a copy?
78 series regulators require the input voltage to be around 2v higher than the output to do anything so it doesn't make sense to claim a 5v supply (superficially at least).

dlmock99, 2022-12-01:
It's the real deal.  I agree with you that the supply voltage should be higher that the regulated output.  The encoder came with the data sheet and it specs. that the supply voltage is to be 5 to 24 volt DC . Yet at 5 volts there was no output.  Removing the regulator and adding a jumper will get it working.
[image]

closed 2022-12-01 by dlmock99