clough42/electronic-leadscrew #240: Need help debugging - spinning encoder does nothing to Stepper

bgrimwade, 2022-12-01:
HI All,

I'm a complete novice when it comes to Steppers etc, but I'm a fair way through getting everything set up, but have no connection between spinning the encoder and seeing the stepper spin.

I've got the boards soldered and connected, the encoder connected to the TI board and the controller also connected. When I spin the encoder I can see the RPM on the controller, so I guess these are connected correctly. I've loaded the latest release build of the firmware on to the board and set the pins on the TI board as well. 

The kit I'm using:

3NM Nema 23 Closed Loop Stepper Motor 4205z - Hybrid
CL57T Closed Loop Stepper Driver
Omron E6B2-CWZ6C Encoder

I've double checked the soldering and followed the wiring diagram I believe correctly. I'm not able to post images of the setup of the wiring here, but do have a build thread on a forum [here](https://www.hobby-machinist.com/threads/restoring-hopefully-my-new-smart-and-brown-sab.100937/page-9) if anyone is able to look?

The only difference in wiring I can see from the documentation on this site, is that the image shows FLT+ and FLT- on the driver, and I don't have those. I was suggested to use ALM for FLT+ and COM for FLT-. Other than that, I think the wiring is correct per James diagram. 

Any help on how to debug this further would be very much appreciated.

Chucksbp, 2022-12-01:
Being that your getting an RPM reading on your display it sounds as though you are wired properly to that point. when you turn on the power to the motor does it lock up the motor?? Motor should be able to be turned when everything is off but should hold when powered up. Do you have the dip switches in the proper off and on settings? Did you check all the continuity on the board for the output side of the Board?

bgrimwade, 2022-12-15:
> Being that your getting an RPM reading on your display it sounds as though you are wired properly to that point. when you turn on the power to the motor does it lock up the motor?? Motor should be [...]

Thanks for the reply. I’ve just double checked the dip switches and run through continuity checks again, and all seems ok. 

The motor locks up when powered on and I can spin it when power is off. 

I can only assume it must be the wiring between the driver and the stepper or the Clough42 board and the driver. I’ve double checked it all and am out of ideas. 

I really have no idea where to go from here.

Chucksbp, 2022-12-16:
Unfortunately I'm far from an expert, but I do have a bit of electronic background, I would say you are ok with the motor to control wiring being that the motor locks up when powered. I would concentrate on the enable - circuit to the controller. First is there a dip switch for enable plus or minus on the controller?? you may try reversing that for a quick check. I would remove the Clough42 board and carefully check all the continuity using the back side of the pins, not the soldier joint. checking the soldier may not have continuity to the actual pin,. Then i would check the same from the clough42 board to the controller. 
   All i can say is keep trying, its great when you get it operating. I'm a bit surprised one of the experts haven't chimed in with some help. I wish you the best in figuring out the problem. One of the boards could have been damaged i guess. I had issues with my display at first, it was a bad board or I ruined it changing the plug???? after a new board it worked fine for a short time, then i believe the encoder failed and took out input one on the TI board. after reprograming to input 2 and a different encoder all ahs been fine!

bgrimwade, 2022-12-16:
> Unfortunately I'm far from an expert, but I do have a bit of electronic background, I would say you are ok with the motor to control wiring being that the motor locks up when powered. I would concentrate on the enable - circuit to the controller. First is there a dip switch for enable plus or minus on the controller?? you may try reversing that for a quick check. I would remove the Clough42 board and carefully check all the continuity using the back side of the pins, not the soldier joint. checking the soldier may not have continuity to the actual pin,. Then i would check the same from the clough42 board to the controller. All i can say is keep trying, its great when you get it operating. I'm a bit surprised one of the experts haven't chimed in with some help. I wish you the best in figuring out the problem. One of the boards could have been damaged i guess. I had issues with my display at first, it was a bad board or I ruined it changing the plug???? after a new board it worked fine for a short time, then i believe the encoder failed and took out input one on the TI board. after reprograming to input 2 and a different encoder all ahs been fine!
> 
> by the way: Also look closely at the board for any bridging between soldier connections. you can also check this with your ohm meter when checking continuity,

Thanks for the quick reply. 

There’s no dipswitch that I can see for for enable + that I can see. 

I’ve double checked the continuity on the board several times, but I’m still not convinced it’s correct. 

My issue at the moment is I’ve no way to test individual components and the intergration between just two components to isolate where the problem is. I’d like to figure out if there’s a way to ensure that the soldering is correct, then move on to make sure that the wiring to the stepper is correct, then the encoder etc. 

I’ve been trying at this for months and really am very close to giving up. I feel like I’ve hit a dead end and don’t know which way to go.

Chucksbp, 2022-12-16:
I understand what you are saying, to test anything above continuity and voltage you need more sophisticated tools and above all electronic knowledge. I ran into this with the encoder issue. and if you do find someone that has the knowhow they speak over your head! Im 73 years old my biggest problem is my hands are getting shaky and my eyes aren't the best. I limited my time when doing the finer points. If you want . Are you absolutely sure of the switch positions on the TI board. they can be very deceiving. you may want to try switching them back and forth a couple times and make sure of the proper position. Also someone was having an issue and he reloaded code composer and the software and re programed the TI. fixed the problem, you may want to just re program the board??? 
   Also If you care to go through the trouble, take some closeup pictures of all your components wiring etc. you can Email them to me or use Messenger on face book. Im retired and have time to look at what you have. mayerchak@comcast.net   facebook Charles Mayerchak Ill try to help.

Chucksbp, 2022-12-17:
I was thinking some more about your problem, I know you said the display shows RPM, does it also show the 4 LED s on and display the cut value?? Can you then change to threads per inch and forward and reverse etc. 
  I have an emergency stop and a regular stop added to my set up. all it does is interrupt the enable wire from the controller. so today i put it in emergency stop to see how the motor acts. the motor stays locked up it doesn't free wheel as if you turned it off. When I was figuring this out initially nothing else stopped the motor. so my thought if all the buttons on the control panel change what they are intended to change the problem has to be in the enable wiring. 
[image]
[image]
This is what it should look like after the startup sequence.

bgrimwade, 2022-12-18:
Thank you again for your help Chuck. I posted this problem on another forum and managed to get the answer there. The stepper controller has a 5V/24V switch which by default is on 24v. It needed to be on 5v for this setup. All seems to be working fine now (on the bench at least). 

I’d be interested to see pictures of your final setup if you have them? I’m looking for inspiration about how to have all of the componentry enclosed and how to mount the stepper/encoder.

Chucksbp, 2022-12-18:
I am so glad you got the problem solved! the ELS is such a great system when working! it really makes a great improvement over change gears. 
My installation is on a very old small craftsman lathe, same as an Atlas 618. I removed all my gears and installed a 60 tooth pulley directly to my spindle shaft. and mounted the encoder, with a matching 60 tooth pulley, for the 1 to 1 ratio, to the gear banjo.  I had previously extended my lead screw on the tailstock end for a dc drive. so I used that to connect the motor at a 3 to 1 ratio. 20tooth on the motor 60 tooth on the lead screw. If you browse through the issues both open and closed you will find a few pictures also that may help you. I wanted to keep the integrity of my old lathe so i wanted the encoder out of site and under the gear cover. 
[image]
[image]
[image]
[image]
[image]
I got my pulleys from ebay and Amazon although later i did find some US vendors.

bgrimwade, 2022-12-20:
> I was thinking some more about your problem, I know you said the display shows RPM, does it also show the 4 LED s on and display the cut value?? Can you then change to threads per inch and forward [...]

That’s a great looking setup. Your lathe looks similar in size and age to my Smart and Brown SAB. I’d thought about fitting the lead screw pulley at the tail stock end, but the lead screw isn’t visible through the casting that holds it. How did you drill the lead screw to attach the pulley?

Chucksbp, 2022-12-20:
Why don't you email me rather taking up room on this site, I actually extended my lead screw, very easy to do. I can explain in detail with pictures. Send me an email and Ill get back to you. my email is  mayerchak@comcast.net.

Chucksbp, 2022-12-20:
I decided to post it here, if they don't approve they can remove it, possibly it will help someone else. 
  My lead screw is 1/2 inch 16 the end where it goes into the bearing is 3/8  I started by machining a half inch piece of round stock down close to 3/8 i left it about .040 large. I drilled and threaded the end of one side 1/4 20 and had the other side center drilled. I then removed my lead screw and chucked it up in a 4 jaw chuck, ijust had the end sticking out of the chuck. I got it centered as close as humanly possible. I faced the end and drilled and taped it 1/4 20. I then used a piece of all thread actually the threads from a bolt. I put Green Loctite {sleeve and seat setter} and tightened them together. let it set for 24 hours and then finish machined the extension to 3/8  I mad a small bronze bearing for extra support although I don't
 think that was at all necessary. 
[image]
[image]