clough42/electronic-leadscrew #241: Alter feed speed by 50 percent

Aditchf, 2022-12-22:
Hello All,
ELS is working perfectly at the moment but would like to reduce the feed by half without affecting the thread pitch. Any help world be great.
I’d also like to add 6tpi to the imperial thread values, it currently only goes to 8 tpi. As you a probably aware I’m a complete novice regarding coding.
Thanks in advance
Andy

tomvikse, 2022-12-22:
In Tables.ccp
To get 6TPI:
Add this line  { .display = {BLANK, BLANK, BLANK, SIX}, .leds = LED_THREAD | LED_TPI, TPI_FRACTION(60) },
over this line
{ .display = {BLANK, BLANK, BLANK, EIGHT}, .leds = LED_THREAD | LED_TPI, TPI_FRACTION(80) },

Aditchf, 2022-12-22:
Wow, thanks so very much. Have a happy Christmas and a happy New Year. Never posted anywhere before and this has been a great first experience.

Aditchf, 2022-12-22:
Hello again. 
regarding halving the feed I don’t think I was very clear in the request for help. I need to alter the ratio by 50 percent. Currently I’m setting 0.1mm per rev and when I rotate the spindle 1 revolution I get 0.2mm of physical movement on the lathe bed. I have a separate feed lead screw. Does what you sent earlier resolve this? I really sorry to ask

tomvikse, 2022-12-22:
> Hello again. regarding halving the feed I don’t think I was very clear in the request for help. I need to alter the ratio by 50 percent. Currently I’m setting 0.1mm per rev and when I rotate the [...]

nope.

tomvikse, 2022-12-22:
In Configuration.h

// Separate step and microstep settings for feed rates.  Redefine these if your
// lathe has a separate feed drive train with a different ratio.
#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS
#define STEPPER_RESOLUTION_FEED STEPPER_RESOLUTION

tomvikse, 2022-12-22:
like this then
// Steps and microsteps
#define STEPPER_MICROSTEPS 8
#define STEPPER_RESOLUTION 200

// Separate step and microstep settings for feed rates.  Redefine these if your
// lathe has a separate feed drive train with a different ratio.
#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS 4
#define STEPPER_RESOLUTION_FEED STEPPER_RESOLUTION 200

Now feed is 1/2 speed..

Aditchf, 2022-12-23:
Hello yet again,
Adding these values create line errors on all the fractional feeds I’m guessing the numbers are getting too large or a summing error. As I won’t be using fractional feeds( only metric) can I comment the fractional feed out? Fractional threading is really useful to have and I’d like to keep this facility.
Again any help would be appreciated 
Regards
Andy

tomvikse, 2022-12-23:
post config file

Chucksbp, 2022-12-23:
What value did you originally have in steps and micro steps

Aditchf, 2022-12-26:
Hi,
Threading is set 3:1, 3 micro steps and 1000 steps (servo drive). To reduce the ratio by half I entered 1 micro step and 1500 steps to keep whole numbers. I’m also getting a communication error which may or may not be related, this was saying I needed a firmware update. The boards were ordered a couple of years ago from James and one of the video states that if ordered from him I would already be firmware release 2. Code composer studio is 9.0004 I believe, should I load CCS 9.0000?

Aditchf, 2022-12-26:
When reflashing, I leave the board connected to everything (5v supply to the board, display connected and outputs to the servo) and just plug in the USB. Should these be disconnected?  I think when James was programming the board he only had the 5v supply and display connected.

Chucksbp, 2022-12-26:
for the first question the stepper resolution has to stay the same, it has to be what the motor dip switches are set to. so if you have the motor resolution at 1,000 then both settings are 1,000 this is causing the error message i would think. if your threading is set at 3 to 1 ratio same as mine then for threading you should have 1,000 resolution and 3 microsteps. if you want to slow the lead screw by half you now need a 6 to 1 ratio, in the feed micro steps you will want 6 and the resolution will stay at 1,000 that would slow the lead screw by half. 
  To your second question I have my code composer on my home PC my shop is about 200ft away from the house so i have removed the board taken it to the computer reinstalled the three jumpers and just used the supplied cable to program my board, no outside power source used. That being said, if you have the ability to take code composer to your machine, i see no reason that it should not work as is. I have a programer for my truck and all I do is plug it in install an auxiliary battery charger to maintain proper voltage because it is a much longer process and reprogram away, you don't disconnect any sensors or actuators. Its the same process in my mind only a huge program comparatively. 
  I hope this helps.

dawesy72, 2022-12-28:
Sorry to jump in but I'm having a similar issue. My feed screw needs 26 turns to move the hand wheel one full turn but the threading lead screw only needs 13. 
No matter what values I put in the feed stepper resolution I get errors in the tables.cpp (const FEED_THREAD inch_feed_table)
Even setting the feed microsteps to 3-1000 as per the lead screw gives the same error when building. Leaving no values in the separate feed microsteps allows it to build correctly but the feed is too slow.

tomvikse, 2022-12-28:
I did a typo
#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS 4
#define STEPPER_RESOLUTION_FEED STEPPER_RESOLUTION 200

do it ike this..
#define STEPPER_MICROSTEPS_FEED 4
#define STEPPER_RESOLUTION_FEED  200

Chucksbp, 2022-12-28:
You actually need to go twice as fast another words a 1.5 to 1 ratio. from my understanding fractions aren't allowed, can you try leaving the 3 microsteps and changing the resolution from 1000, to 500 that would give the same outcome of 1500 pulses per revolution. that's if the software will allow this change. 
I believe the trick may be on line 71 and possibly even line 70 but id try line 71 first. remove the //  and add  #define   This line may then allow you to put in and accept the second set of values.

dawesy72, 2022-12-28:
That works a treat. Need to tweak the settings a little but now have way better speeds. Thank you.

Chucksbp, 2022-12-29:
> That works a treat. Need to tweak the settings a little but now have way better speeds. Thank you.

  Can you please explain exactly what you put where in code composer to make things work that way we can help someone else in the future. It rare there is any experts on here lately it seems.

dawesy72, 2022-12-31:
So in the line for FEED_MICROSTEPS there is an error 

typo
#define STEPPER_MICROSTEPS_FEED STEPPER_MICROSTEPS 4
#define STEPPER_RESOLUTION_FEED STEPPER_RESOLUTION 200

do it like this..
#define STEPPER_MICROSTEPS_FEED 4
#define STEPPER_RESOLUTION_FEED 200

Remove STEPPER_MICROSTEPS and STEPPER_RESOLUTION

If you remove the above it will compile and work.

Chucksbp, 2022-12-31:
Thanks so much!! i have a friend that will need to do that very thing soon, i'm sure we would have run into problems.

dawesy72, 2022-12-31:
No problem. Mines working great at the moment though I may have to drop my stepper to screw drive as the motor can't keep up at the highest speed and feed combo.