clough42/electronic-leadscrew #242: No RPM displayed, 5vdc pulses present, tried both encoder inputs

oracleglassarts, 2023-01-01:
No RPM displayed, 5vdc pulses present from all three putputs, tried both encoder inputs.  I don't know what else to try besides swapping parts.

I'm using the "omron" encoder linked in "stuff you'll need."

I also double checked the jumpers and switches that need to be set on the TI board.

Please help if you can!

Chucksbp, 2023-01-02:
Did you reprogram the launch board to use the second input?? hAS the system ever worked or is this a first start up?? I would double triple check the soldier joints on the two large plugs that connect with the launch board. If so I would first try the encoder They seem to be troublesome??? I'm not sure that you can properly diagnose them with out an oscilloscope??? Then I guess the launch board??

oracleglassarts, 2023-01-02:
Yes - I did swap outputs by commenting those lines in/out of the config file.  I built a MPCNC a couple summers ago so all the firmware flashing stuff is familiar to me.

I checked for shorts between pins when I soldered the PC board and everything looks good on the solders (I've built a few synthesizers from kits - so through hole soldering is another familiar thing to me.)  I also used a brand new soldering iron with temp control so these are probably my best solders ever.

My programmer buddy is saying to check the encoder with the meter on AC - I'm not quite sure if/how that will work but im.going to try that next.  He's starting work for the same company I work at tomorrow so we can get together and look at it there

Chucksbp, 2023-01-02:
My attempt worked well right from the beginning but failed within a few hours of use, as did another gentleman below. we both had the same issue at the same time. We both purchased a different brand of encoder from Amazon, both reprogramed to the second input and had working results that has lasted for very close to a year now. Neither of us tried just a new encoder or the old encoder on the second input, although I do believe someone just changed one, i don't remember positively which way. I was afraid of damaging the second input??? There seems to be some failures with the eBay encoders though. After that I put together a second working and tested spare ELS less the Motor for spare parts. The motor and controller are the only more expensive part. Good luck with your trouble shooting! Its a great improvement to a lathe, you will enjoy it!

oracleglassarts, 2023-01-02:
Thanks, I'll probably end up trying an Amazon encoder.

Mouser has quite a few Launchpads in stock, 1,149 to be exact 😀 I'm really hoping I didn't fry both inputs on mine with a bad encoder - knowing my luck that's probably what happened though lol.

Chucksbp, 2023-01-03:
Good luck!!! this is the encoder we used.    https://www.amazon.com/gp/product/B08RS6M32J/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1

oracleglassarts, 2023-01-07:
[image]
[image]
[image]

closed 2023-01-07 by oracleglassarts