clough42/electronic-leadscrew #245: Suggestion: Manual Jog/Stepping

Grauenwolf, 2023-02-15:
Normally this turns the leadscrew in response to the spindle, but sometimes that's not what we need.

I've seen projects where we have to move the carriage over a specific amount, cut with a form tool or parting blade, then move and repeat.

An example of this is a Blondehacks video where she is roughing a ball or other curve. 

Thus it would be useful if we could switch the control into 'manual' mode where the up/down button would move the carriage a pre-set distance.

kwackers, 2023-02-16:
So my thoughts:

The biggest issue I think is the user interface, how would it work?
The problem with a "pre-set" distance is how much should it be? 1 thou? An inch?
Too small and you'd be there forever, too large and it'll be useless for most folk.

That means you need some mechanism to set that value. In the custom thread mod I did you could change digits with the set and up/down buttons which could be used to set a value but then there's no button to move the carriage.
You *could* switch between setting the jog value and moving the carriage but there's no obvious button for that and so it could get a bit clunky.

Given the relationship between the spindle and the carriage you *could* move the carriage by turning the spindle - all that's really missing is a carriage spindle display.
So you could put it into 'manual' mode which switches the display to position, the up/down buttons could then set how course the movement was and position it by turning the spindle.
Not ideal but.
(Even better would be another encoder for moving the carriage).

OG1-SS, 2023-02-16:
I have this setup that does this perfectly on an import conversion. I wish
I knew how to read the code off it to understand and share. I have been
waiting for someone to come up with a similar method so I can use the two
clough units I have. Mine lets me input pitch then distance in mm then
choose between auto or manual so I choose manual and it pauses at the end
of the preset distance (where it is safe since I made the relief there)
then I back off the X and click return and it starts the cycle over, I
always allow a gap at the beginning of the cycle in case I need to pause
it, I would prefer that part be manual as well.

Grauenwolf, 2023-02-16:
@kwackers Yea, I'm thinking that it would need a separate button or at least mode indicator light.

I understand why he chose physical buttons, but this screams "if only I had a touch screen".

Grauenwolf, 2023-02-16:
@OG1-SS That is different, but a heck of a lot more sophisticated, than I was thinking. It should have its own suggestion.

kwackers, 2023-02-17:
@OG1-SS isn't that just the same as threading to a shoulder though?
I added code for that in my branch, the only difference is I don't take a distance, instead you start the carriage at the place you wish to stop (the shoulder) and then move it to the start position.

When you thread the carriage automatically stops at the shoulder, you retract the X and press the up button at which point it moves the carriage back to the start position again.

OG1-SS, 2023-02-17:
Oh man, I missed that. Thank you for letting me know. I can't wait to try
it. I'm new to this.

tomvikse, 2023-02-17:
«threading to a shoulder» Works perfect, making threds @ 800rpm up to a shoulder no problem.. 😀 stops at 0.000 every time on the dro