clough42/electronic-leadscrew #246: Encoder not talking to servo controller flickers

deirpo, 2023-02-19:


Chucksbp, 2023-02-19:
Does it show RPM on the display?? The display boards thyem selves are quite finicky. My first one didn't work properly. you can buy them on Amazon, quite inexpensive. Try a new one without changing the connector. 
Also there have been a few non working and early failure ebay encoders. there are a few issues posted below. also has been some TI launch board failures, nesetating changing over to the second Launch board encoder input. I had an early Encoder failure. I replaced the encoder with one from Amazon and switched to the second input and it has been fine for about a year now.

deirpo, 2023-02-19:
Hi, No rpm showing when encoder rotated either direction and no servo motion. When I power up the display shows the complete sequence of boot up and all buttons working and all modes responding properly to button pressed. The digits are not bright as they should be and there is random flickering.   I’ve ordered a couple new displays already.Thanks for your response Barry Sent from my iPhone

deirpo, 2023-02-19:
Hi,   Just received new led8key, this one works perfectly! 👍 Now to install connector on back.   Still no function from encoder :(  Can I test encoder with oscilloscope? I’ve never tried, does anyone know how to?Regards Barry Sent from my iPhone

Chucksbp, 2023-02-20:
I am sure the encoder can be diagnosed with a scope although I'm not sure what you would expect for a good one. but I do believe there were some videos on line.
 I was able to leave the conector where it is, bend the pins at a 90 degree angle trim a bit of excess shell on the plug and still get it all in the box. I was afraid i may have ruined it removing the old pins. later I did manage to change the pins and it worked. I have spare parts just in case.

deirpo, 2023-02-20:
I replaced the original display keyboard and the TI board boots up normally and no more flickering! Ordered a replacement encoder, should be here tomorrow. Hopefully it won’t be dead on arrival, too!Barry Sent from my iPhone

deirpo, 2023-02-28:
Solved! Bad encoder and lednkey panel. Go figure!

Chucksbp, 2023-03-23:
https://github.com/clough42/electronic-leadscrew/issues/246#issuecomment-1480478130
  How are you driving your encoder?? My suspicion would be a slight offset in the 1 to 1 drive ratio. Mark the position of the spindle and the encoder. run the lathe for 5 minutes or so and see if the two markes still line up. I had an issue with Chinese pulleys. same number of teeth but the encoder would gain tining when running for a while. I also found belt tension on the encoder to be  very touchy. 
[image]

dadyo55, 2023-03-23:
Thanks the encoder are to pulleys I printed both the same ? But I will run them for 5 minutes and see if they are still timed right

Chucksbp, 2023-03-24:
I have no idea of how to properly test the encoder although i believe i once found a youtube video but you needed an oscilloscope for proper diagnosis. this is the encoder a couple of us used when we had failures. I believe the Chinse knock offs on ebay are having some issues at times.  This one seems to be working well. 
  https://www.amazon.com/gp/product/B08RS6M32J/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1

deirpo, 2023-03-27:
Thanks Chucksbp for the feedback. I took your advice and order the encoder you suggested. Seems to work right out of the box! 
   Now if I can just program my servo and launchpad with the correct micro steps to control feeds on my G0602 lathe with 3 to 1 step down pulleys and 2 to 1 gear reduction.
   Thanks for your help!
Barry

Chucksbp, 2023-03-27:
You are very welcome! it seems as though that encoder is better built than the ebay ones. You are the third Pearson i'm aware of that had success with it with the ELS. 
  I believe if you set your resolution to the motor setting and the microsteps to 6 it should work. As long as the total doesn't exceed the limits of the program. if so you can lower the resolution on both the motor and the program.