clough42/electronic-leadscrew #247: PM-1130 - feed shaft configuration?

javaduke, 2023-02-20:
Folks, I'm trying to figure out the settings for my PM-1130 lathe. The spindle to encoder and motor to shaft ratios will be 1:1 so no problems there, the lead screw is 8TPI, also pretty clear. I'm planning to leave my internal gearbox at the `B` setting so it will be 1:1 ratio as well. But what I cannot figure out is the power feed settings. I experimented with the  feed shaft and found that the saddle moves 0.08" per each revolution of the feed shaft. So the question is how does it translate to the `STEPPER_MICROSTEPS_FEED` and `STEPPER_RESOLUTION_FEED` settings? 
Pardon my ignorance, I still consider myself new to the machining, especially the lathe works.

tomvikse, 2023-02-20:
Pully size on servo?
Pully size on the gearbox?

javaduke, 2023-02-20:
I will use gears, not pulleys - and the ratios will be 1:1 for both spindle to encoder and servo to shaft.

tomvikse, 2023-02-20:
ok 
Think running 1:1 is going to give you problems.. servo is going to me to weak.. i run the same servo @ 6:1

did you change the command puls count from 4000?
https://youtu.be/clgpZJFoqIQ?t=668

javaduke, 2023-02-20:
No, I haven't changed any settings yet, but planning to.

kwackers, 2023-02-21:
0.08" per rev is 12.5 tpi.
Since your leadscrew is 8 tpi then the feed is running at 0.64 the rate so you need to fudge STEPPER_MICROSTEPS_FEED and STEPPER_RESOLUTION_FEED values so they give you that 0.64 reduction without resorting to floats.

So if for example your current value for STEPPER_RESOLUTION is 200 and STEPPER_MICROSTEPS is 8 this gives you a total of 1600 steps per rev of the motor, to move the same distance using the feed would need 1600 / 0.64 = 2500 steps so you could for example get this by using 625 and 4 for the two values above.

Make sense?

(Haven't looked at the code for a while but hopefully I haven't forgotten how it works... ;) )

javaduke, 2023-02-21:
Ahh, ok, now it does make sense, I got confused because feed shaft is not a screw really, but thinking of it in terms of TPI actually does make a lot of sense. I was also thinking about adding a new definition in the Configuration.h, something like `#define POWER_FEED_INCH_PER_REV ...` but your solution is even cleaner. Thank you!

closed 2023-02-21 by javaduke