clough42/electronic-leadscrew #248: 36 vs 48 volt power supply?

javaduke, 2023-02-26:
I'm a bit confused, I ordered the servo that James suggested, and it says 24-36 Volt, but the power supply is 48 volt - so is it ok to cover the servo with 48 volt instead of 36, or should I get the 36 volt power supply?

tomvikse, 2023-02-26:
48V no problem..
https://youtu.be/clgpZJFoqIQ?list=PLDlWKv7KIIr90ZZ7Zqt-ge5nVVdS3WVgg&t=150

javaduke, 2023-02-27:
OK, thanks, hopefully it won't burn down :) It is confusing though, the product description on StepperOnline says 20-50VDC, the label on the integrated controller says 24-36V and the power supply is 48V...

closed 2023-02-27 by javaduke

johnsonm, 2023-04-08:
I note that James had his first servo die, and 48V is close to the rated maximum; that's not a lot of headroom for transients, say from back-EMF. The servo's rated torque is at 36V, so it's not crazy to run it on 36V, with more headroom.

I will be running my servo at 36V for this reason, and if I swap out for 48V due to insufficient torque, I'll plan either to adjust the nominal 48V supply a few volts lower, or to add transient protection; at least a large-value capacitor, and maybe a crowbar shunt to avoid exceeding 49V. But I don't expect that to be necessary.

tomvikse, 2023-04-08:
Using a 48v power supply adjusted down to 42v.. good for 2 years now..