clough42/electronic-leadscrew #249: Problems with power feed

javaduke, 2023-02-28:
First of all I really want to thank James and everyone who contributed to this project, it's really awesome! I finally got all the components, programmed the motor and the launchpad and everything is working correctly, the control panel shows the correct RPMs and the motor is spinning! Yay!

Now to the problems. First, in the threading mode the motor is spinning in the opposite direction, feeding the carriage away from the chuck. I selected Rev on the panel and it started feeding towards the chuck as it should for the right hand thread. I tested several imperial threads and they look great! So aside from the fact that in forward mode it feeds in reverse, everything is ok.

The problem now is with the power feed. In the power feed mode the motor is spinning and rotating the power bar as it should, but the moment I engage the feed, the motor stops and the alarm is blinking 4 times (Overload Alarm). So I'm wondering what could possibly cause this and why is it not happening in threading mode.

Now, my ratio from the motor to the shaft is 1:1 and in the internal gearbox I selected 0.5:1, that is, the overall ratio is 2:1. Wondering if this is enough or if I should try 2:1 to the shaft and 0.5:1 in the gearbox.

kwackers, 2023-03-01:
You can define INVERT_DIRECTION_PIN in Configuration.h to flip the direction - or you can simply pick one pair of wires for a single motor winding (A or B) and swap them.

Do you engage the feed whilst the stepper is spinning? If so you could be stalling it and stepper motors can't usually recover from that.
If not then perhaps the feed is stiff, or has higher overall gearing?

Vibrating or pulsing whilst actually running or when it's supposed to be stationary?
On mine I found some vibration when it should be stationary (in all modes) so I added electronic backlash of a couple of steps to stop this (see my fork).
If whilst running then it sounds like the motor is struggling and on the verge of stalling, if you apply a bit of load does that stop it completely?

javaduke, 2023-03-01:
What I meant is that in the power feed mode the motor is rotating but not smoothly, it looks like it's jerking or pulsating. I'll try to make a video later today. Yes, it seems like the motor is stalling, and yes, if I engage the feed it stalls completely. I ordered some more gears and will try different ratios to see if it improves the situation.

Thanks for the INVERT_DIRECTION_PIN tip, I'll give it a try.

javaduke, 2023-03-01:
Here's the little video I created: https://www.youtube.com/watch?v=LMWwvt6kDCM (I apologize for the quality of the video, I'm not nearly as good at it as James and many other YouTubers :))
Oddly enough there's a positive change from yesterday, both lead screw and feed shaft are now rotating - but you can see how jerky the gears are. Looks like the motor is struggling, especially at slow speeds. I suppose I need to have 1:2 or even 1:3 gears from the servo to the shaft and leave the internal gearbox at the C setting (1:2).

tomvikse, 2023-03-01:
Is it the 180w servo from stepper online?

javaduke, 2023-03-01:
Yes, the same one James linked in the Wiki: https://bit.ly/37SctTB
I downloaded the latest version of the software and changed the Microstep parameter from 4000 to 1000, but didn't change any other param.

tomvikse, 2023-03-01:
You need to run it at 6:1

javaduke, 2023-03-01:
Understood, thank you! Hopefully the new set of gears arrives soon.

tomvikse, 2023-03-01:
3:1 can work but 6:1 and you are safe

peterdownunder, 2023-03-10:
As an aside what power supply are you using to drive the steppers

javaduke, 2023-03-11:
The same power supply that James recommended on the Wiki page. I ordered it directly from the StepperOnline, with the servo.

tomvikse, 2023-03-11:
> Understood, thank you! Hopefully the new set of gears arrives soon.
> 
> Should I also increase the Stiffness parameter or I'd be better off leaving it alone?

I left the Stiffness parameter as it was..  I have run it very hard and it just works.. same hardware as you 6:1

closed 2023-03-14 by javaduke