clough42/electronic-leadscrew #251: Error: Program "make" not found in PATH

KX31, 2023-03-30:
Hello all,

I am trying to load the code to the launchpad but i am experiencing some error trying to build the code.
I started with CCS V12 , then CCS V10 and V9; they all have the same issues. 

Am i missing something in the install or setting?

Here is the error message:

{time} **** Clean-only build of configuration Release for project els-f280049c ****
make clean 
Cannot run program "make": Launching failed

Error: Program "make" not found in PATH

PATH=[
... ... ...
]

{time} Build Failed. 1 errors, 0 warnings. (took 48ms)

**** Build of configuration Debug for project els-f280049c ****

Nothing to build for project els-f280049c

Thanks for your help,

clough42, 2023-03-30:
The firmware is written for v9.  If you're still seeing the error, it's
likely your system is still trying to use V12 or some newer version.  I'd
probably start by making sure everything is removed from your machine and
reinstalling v9 from scratch.

James

KX31, 2023-03-31:
Thanks for the help, i managed to un-install V10 & V12 properly and re-install V9.
It build properly, but now it error to load onto the board.

I went back over the video and spec sheet and something is wired:

The board select properly:
[image]

But it doesn't have the XDS degugger in the selection:
[image]

And explicitly state the XTS is not installed:
[image]

Im missing something (that im sure is simple) but im stuck.

Ho, and here is the error code on loading: 
[image]

I tried installing V9 a few times, and always get the same selection on the install procedure (always missing the XDS)
Is there a dedicated install needed for it?

KX31, 2023-04-04:
I re-installed V9 with all the board and options and the XDS110 debugger is not supported either.
Do you have a source file for the V9 that you could share to test?

closed 2023-04-10 by KX31