clough42/electronic-leadscrew #252: Boost board provided without flash chip

johnsonm, 2023-04-12:
The flash chip was not included on the boost board I purchased. A note accompanying the shipment noted that this was due to the chip shortage. This wasn't noted anywhere before I bought it.

I know it's not currently used, but I had considered adding functionality that would use it, and was disappointed that it wasn't included.

If this had been made clear in the ebay page or in the documentation here, I could have added the $0.44 part to my mouser order of the launchpad, as mouser has plenty in stock.

Now I guess I'll wait for my next mouser order and hope it's still in stock then.

It would be nice to include the information that the part has been dropped from the BOM both in the project documentation here on github and on the ebay page, for those of us who have an interest in making use of it.

meckarmulle, 2023-04-13:
Hi
I had some problems with my built and posted an issues, and the missing chip was speaking.
https://github.com/clough42/electronic-leadscrew/issues/223

It was not the reason for my problems and my built works perfectly fine now, I use it almost every day.

BTW, I like your new post and follow it 
BR
Perry

johnsonm, 2023-04-13:
I must have found a similar-looking one; either that or they were snatched up. Whatever I saw had 700-odd in stock when I looked, and I think was $0.44 for onesies.

But it would hardly take 16kb / 2kB to store, say, some power-on defaults. For example, this 2kb / 256B flash looks pin-compatible and would have added $0.35 to my mouser order for the TI dev board without additional shipping. I hate paying $8 to send a spec of dust halfway across the country... ☺

https://www.mouser.com/ProductDetail/Microchip-Technology/AT25020B-SSHL-B?qs=fzsaQExJLyQk1UjUU2jjPA%3D%3D

In any case, I just wish I'd known ahead of time, since I was actually planning on using it.

clough42, 2023-04-13:
Fair enough.  It would be worth adding a note to the eBay listings.  I hadn't considered that people might be writing their own firmware to use it.  I was actually thinking about removing the footprint from future boards, but maybe it's better to leave it there in case someone wants to use it.

I'm sure you're aware that if you use the chip you linked above, the page size and address encoding are different, so the code in EEPROM.h and EEPROM.cpp will need to be modified.  There are two different EEPROMS supported at the moment, but I think the one you linked uses yet a third addressing scheme.  I would have to spend more time with it to be confident I'm reading it correctly.

It's a little bit maddening because they make a bunch of different chips that are all very similar but differ in important ways.  Some will read out 32 bits sequentially.  Others will loop around and return the same 8 or 16 bits repeatedly if you try.  Some require the high bit(s) of the address to be encoded into the read command opcode, and others have a full 16-bit address following the 8-bit read opcode.

closed 2023-04-13 by clough42

johnsonm, 2023-04-14:
Yeah, I was figuring whatever I used I'd be starting from the datasheet. But thanks for the heads-up for sure. I'll see whether I miss the abilities I was planning for it in practice before placing an order (or... scrounging my dead boards for parts)

johnsonm, 2023-04-14:
FWIW, for anyone else who finds this and is looking, here's a 32kbit part currently $0.84 for onesies that looks pin-compatible, and from the datasheet doesn't look too hard to modify EEPROM.cpp to handle.

https://www.mouser.com/ProductDetail/ROHM-Semiconductor/BR25S320FVM-WTR?qs=IsRgwgmxh6%252Bb2WodExhY9g%3D%3D