clough42/electronic-leadscrew #259: Inaccurate thread pitch

oldedwardian, 2023-05-01:
Hi All,
 I am a complete novice at coding, and I have only got this far by following James’s instructions to the letter. Any help would be gratefully received.
I am using all the equipment James has suggested and changed the setup parameters as he suggested in configuration h. and 500 microsteps in the servo.
I am using pullies at 3 to 1 (20 teeth on the servo and 60 teeth on the leadscrew pulley)and through my gearbox at 2 to 1 so 6 to 1 from the servo to leadscrew.
My leadscrew pitch is 3mm.
In config.h I have set stepper microsteps 6
Stepper resolution 500
If I set leadscrew to 300 the pitch of the thread being cut is not accurate.
I have improved the 1.75mm pitch being cut to be 0.1mm error over 50mm length of thread by changing my leadscrew value to 292. I cannot get it better than this by this method. 
I assume the error is caused by the Chinese pullies not being spot on and any error on the servo pulley is magnified 6 times at the carriage.
Is there a way of finer tuning this error out in config.h or should I just make some more accurate pullies?
Thanks for any help.

tomvikse, 2023-05-01:
Looks right .. are you sure the gearbox setting is 2:1?  Try 1:1 and see if it is correct then(set stepper microsteps 3)

oldedwardian, 2023-05-01:
Thanks for the suggestion, I'll have another look tomorrow and do as you say.

oldedwardian, 2023-06-02:
I have found the problem. My encoder has 1000 p/r, I thought it was the same as James encoder at 1024, hence the error.
All good now, threads are accurate over several inches.

closed 2023-06-02 by oldedwardian